gsohler/openscad #4: Allow editor to get current python state

Matthieu-LAURENT39, 2023-07-29:
The text editor currently has no way of knowing if python is currently active (i.e. python is enabled and the current file is a python file).
If the editor had that knowledge, it would allow to add much better editing features.

Having a `python_enabled` public attribute on the ScintillaEditor class could do the trick, however i'm not too sure how python mode is actually toggled (it seems like it's computed from the file name every time?), so i can't really add it myself

Matthieu-LAURENT39, 2023-07-30:
I implemented this by allowing ScintillaEditor to access MainWindows's python_active attribute, however i didn't get time to test it, and compilation is taking too long.
I'll test it tomorrow and submit a PR.

gsohler, 2023-07-30:
remarkably compilation is only fast 1st time and slow later.
to get make the boost, do
make clean
make

Matthieu-LAURENT39, 2023-07-30:
It's the other way around for me, it's really slow the first time, then fast since it only recompiles the changed files

Matthieu-LAURENT39, 2023-07-30:
This is implemented and works, i'll make a PR

Matthieu-LAURENT39, 2023-07-30:
resolved in #6

PR #6 (2023-07-30): Adds python style comments to editor
This adds "#" comments when python mode is enabled (resolves #5 )
And most importantly, it allows the editor to know when python mode is on (resolves #4 )
closed 2023-07-30 by Matthieu-LAURENT39