gsohler/openscad #10: Hull() doesn't work in Python mode

Matthieu-LAURENT39, 2023-09-07:
**Describe the bug**
I'm translating this code into python:
```
hull() {
    circle(5);
    translate([0,20,0])
    circle(5);
}
```
The translation:
```py
hull(circle(5), circle(5).translate([0,0,5]) )
```

The original scad code works fine, but the python code raises "TypeError: error duing parsing advanced csg".

**Expected behavior**
Render is successful, and works the same way as regular scad

**Code reproducing the issue**
```py
o = hull(circle(5), circle(5).translate([0,0,5]) )
output(o)
```

**Environment and Version info (please complete the following information):**
 - OS: Manjaro 23.0.0 Uranos
 - System: Intel PC 64-bit
 - OpenSCAD Version: built from source from commit db620c83042b62e3e6bd1120c38b1618d9b10968

Matthieu-LAURENT39, 2023-09-07:
Actually, after more testing, it seems hull() expects a list as input, which caused the error

closed 2023-09-07 by Matthieu-LAURENT39

gsohler, 2023-09-07:
Hi Mattheu,, thank you for pointing out.
I am still not clear, if its better to provide an array with the inputs or if its better to supply the plain arguments.
In any case when providing plain arguments the parameter length is rather fixed whereas when supplying an array we are not fixed to a certain parameter count.

Matthieu-LAURENT39, 2023-09-07:
we should probably move this to #11