gsohler/openscad #11: Improving the function calls message error when a non-list is passed

Matthieu-LAURENT39, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
Currently, if you try to call a function like `union()`, `difference()`, `hull()`, etc with 2 objects as parameter, for example `union(circle(5), circle(5)), you will get the following error: `TypeError: error duing parsing advanced csg`.
The message is really unclear, and doesn't indicate to the user that he should've passed the objects as a list (or even that it's his fault, this message makes it seem more like it's OpenSCAD's fault)

**Describe the solution you'd like**
There are 2 possible solutions to this issue:

### The easy way
Simply add a better error message when multiple such objects are passed to a function like `union()`

### The hard way
Maybe it would be possible to actually make the (currently) erroneous syntax work, using *args?
Using `union(obj1, obj2, obj3)` feels more intuitive than `union([obj1, obj2, obj3])`, although that's just my personal opinion

Matthieu-LAURENT39, 2023-09-07:
I don't mind implementing either solutions, although i'd want your opinion on which one you see fit best @gsohler

Matthieu-LAURENT39, 2023-09-07:
To go back on what you said in #10 

> In any case when providing plain arguments the parameter length is rather fixed whereas when supplying an array we are not fixed to a certain parameter count.

You can use *args so that a function can take as many arguments as it can, and use that

gsohler, 2023-09-07:
I feel, that using a variable argument count for all Boolean operators , hull, resize , etc. is more intuitive, but I did not yet look into ,how variable argument counts  work in python.
Still I want to keep the feature to use [solid1, solid2] as an union in place of an argument.
So 
union([a,b],[c,d] ) 
would be same as
union(a,b,c,d)

For difference it would be a benefit to write
difference([un1, un2], diff3, diff4 ])
because you do not explicitely have to mention that un1 and un2 are additive terms.

If  you know right away, how to proceed,  i would absolutely appreciate.
This would be a change in syntax in very early stage  and ctests would  have to be adapted,  but thats ok.

Matthieu-LAURENT39, 2023-09-07:
> I feel, that using a variable argument count for all Boolean operators , hull, resize , etc. is more intuitive, but I did not yet look into ,how variable argument counts work in python.

It's *really* simple (like most of python :smile: ), it gives you a list of all arguments
For example:
```py
def union(*args):
    print(args)

union(1, 2)
>>> [1, 2]
union([1,2], [3,4])
>>> [[1,2], [3,4]]
```
So with that, it's pretty trivial to extract the objects.
I'm not too sure which file to edit to add that functionality though

gsohler, 2023-09-07:
Had some time to think about it during lunch time.
For functions like scale, translate, which mostly takes non-solid parameters, it absolutely makes sense to to have the solid as the first parameter (and optionally use an array of  solids as 1st parameters). 
This is needed for the oo notation(object  oriented).
Most functions have an oo partner function which is just a wrapper function, which add 'self' to the beginning of the parameter  list and call non-oo function. ( cube.translate() is oo e.g.)

About your message:
I have seen the python *args syntax before, but i hardly used it myself. from your example i see, that it creates an array from the remaining arguments.  This is script side of the function only.
All python related functions can be  found in src/core/pyfunctions.cpp

sample parameter  extraction for translate looks like:

 char *kwlist[] = {"obj", "v", NULL};
  PyObject *v = NULL;
  PyObject *obj = NULL;
  double x = 0, y = 0, z = 0;
  if (!PyArg_ParseTupleAndKeywords(args, kwargs, "OO", kwlist,
                                   &obj,
                                   &v
                                   )) {
    PyErr_SetString(PyExc_TypeError, "Error during parsing translate(object,vec3)");
    return NULL;
  }
  child = PyOpenSCADObjectToNodeMulti(obj);

 for each parameter, you add info in 3 different places: kwlist, type(O) and parameter itself.
I dont  know yet, how to correctly use this with *args  python syntax

Corresponding oo function is just a wrapper function to prevent  code duplication.

PyObject *python_translate_oo(PyObject *self, PyObject *args, PyObject *kwargs)
{
  PyObject *new_args = python_oo_args(self, args);
  PyObject *result = python_translate(self, new_args, kwargs);
  return result;
}

In the very end of the file, all functions features are collected in an array.
This is basic structure.

After all i was wondering why

diff=a.difference(b) 

like mentioned in the tutorial works at all, but I tested and it does work.
python is full of miracles!

(my way of writing would be:)
diff=a.difference([b])
which gets translated to
diff=difference([a,b])

Guenther

Matthieu-LAURENT39, 2023-09-07:
Oh, if it's implemented in C++, then it's a bit more complicated, as i don't have much experience parsing python in C++.
I'll try to make a POC branch later to see how feasible it is to change the parsing logic to accept *args.

On a side-note, i've been making a Python-OpenSCAD script for over an hour straight, and even then i keep forgetting you need to pass a list to `hull()`, the new notation would really be useful

gsohler, 2023-09-07:
The very main intent of my repo is to replace the bison/yacc parser by an embedded python parser.
There are so many python-> openscad language generators out there, and they work quite seamlessly as you can tell openscad to 
re-render when scad code is changed  by the python script, however I am aiming for the direct/native approach.

gsohler, 2023-09-07:
just came into my mind: for boolean ops we can completely skip the PyArg_ParseTupleAndKeywords function because
the arguments are not named. just use the provided Tuple itself.
In case you did not yet start, i think i could "fix" the behavior tomorrow easily

gsohler, 2023-09-11:
i have updated the error message and also the way how the arguments are passed.
please review and close on demand

closed 2023-10-03 by gsohler