gsohler/openscad #15: path extrude a hexagon does not close the top

gandrewstone, 2024-02-10:
**Describe the bug**
there are some triangle gaps when extruding a hexagon up

**To Reproduce**
Steps to reproduce the behavior:

compare py:
output(path_extrude(circle(6, fn=6), [[0,0,0],[0,0,3]]))

with scad:
linear_extrude(3,twist=0) circle(6, $fn=6);

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
ubuntu
commit 2df4f42c6f540b5a1374b86f8976584c66adc051 (HEAD -> python, origin/python)
[image]

gsohler, 2024-02-10:
its fixed!
It was  a stupid bug resulting from too much copy & paste.
thank you for pointing out!

gsohler, 2024-02-10:
your code looks interesting. are you doing a hex bolt screw ?

gandrewstone, 2024-02-11:
yes, I'm creating a test plate to find the right scaling for a really nice hex bolt fit.  I'll update my code soon, but of course cylinder also works... so heading to my printer now :-).  I'm liking how Python list comprehensions work really elegantly here!  And how difference can accept a list of objects...

def hexScrew(shaftDiameter, shaftLength, headOuterDiameter, headThickness):
    hexHeadHeight = 4
    # hex = path_extrude(circle(headOuterDiameter/2, fn=6), [[0,0,0],[0,0,headThickness]])
    
    # circ = path_extrude(circle(headOuterDiameter/2, fn=30), [[0,0,0],[0,0,headThickness]]).color("purple",0.1)
    hex = cylinder(h=headThickness, r=headOuterDiameter/2, fn=6)
    shaft = cylinder(h=shaftLength, r=shaftDiameter/2, fn=32)
    return hex | shaft # | circ

plate = cube([60,10,5]).translate([-5,-5,0.01])
# using ISO 4017:2022 M3 screw dimensions, with a 10 mm shaft
screwSizes = [hexScrew(3,10,6.01,2.125).scale([i,i,1]).translate([i*200 - 200,0,0]) for i in [1.0, 1.05, 1.10, 1.15,1.20, 1.25]]
testplate = plate - screwSizes
output(union(testplate))

closed 2024-02-18 by gsohler