gsohler/openscad #16: pdb.set_trace() causes a hang

gandrewstone, 2024-02-13:
**Describe the bug**

The code below hangs the whole program.  It would be awesome if the console was interactive.  However, that is a big ask.  For now, it would be nice if it pdb.set_trace() became a no-op.

**To Reproduce**

```
# This is a http://pythonscad.org/ file
from openscad import *
import pdb
print("test")
pdb.set_trace()
print("test2")
```

gandrewstone, 2024-02-17:
I think that for now just block the feature.  I think you could create your own fake "pdb" module that would be imported in preference to the normal one and just have it's APIs print something like "interactive debug is not available".  That's better then a hang (esp if you haven't saved your work!)

gsohler, 2024-02-18:
Sorry, I don't have an idea how to create fake python modules which are preferred over the system one's
However, the problem is much more generic: any python script which does not finish within short time, will hang openscad
(pdb will not because it waits for input)
So I prefer a more generic solution instead of creating fake modules.
Don't be so afraid from losing unsaved work. even if you dont explicitely safe before each render, openscad will save your editor content in a backup file.

gsohler, 2025-01-30:
[image]

looks like that it will be possible soon to debug  python code soon
We got an input field below the console and displaying geomtries makes also sense during program execution, not only in the end now