gsohler/openscad #17: Tiny bug: printing a OpenSCAD object chops the rest of the printout

gandrewstone, 2024-02-13:
**Describe the bug**

printing out an openscad object appears to be chopping the rest of the line:

```
prefix None suffix
prefix OpenSCAD (2)
```

**To Reproduce**
try this:
```
import openscad as ops
s = ops.sphere(10)
t = None
print("prefix", t, "suffix")
print("prefix", s, "suffix")
```

gsohler, 2024-02-14:
__repr__ member of openscad object created an extra '\0' character.
Could you check if its fixed in your place and close ?

gandrewstone, 2024-02-17:
got this error building e1f34aab6c77300629b4ae6eb68edfc156813e42
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/geometry/GeometryEvaluator.cc.o: in function `std::ostream& operator<< <__mpz_struct [1], __mpz_struct [1]>(std::ostream&, __gmp_expr<__mpz_struct [1], __mpz_struct [1]> const&)':
GeometryEvaluator.cc:(.text._ZlsIA1_12__mpz_structS1_ERSoS2_RK10__gmp_exprIT_T0_E[_ZlsIA1_12__mpz_structS1_ERSoS2_RK10__gmp_exprIT_T0_E]+0x36): undefined reference to `operator<<(std::ostream&, __mpz_struct const*)'

gsohler, 2024-02-17:
Sorry, cannot See this Error in my Side. Did you try a "make clean" ?

gandrewstone, 2024-02-17:
no trying that now; I was able to build HEAD~1 fine though

closed 2024-02-19 by gsohler