gsohler/openscad #20: Wedge and Arc primitives for python

gandrewstone, 2024-02-19:
Here is some code you might want to add to your python lib:

```
def wedge(radius, angle, fn=None, fa=None):
    """Returns a 2d wedge (pizza slice).  It operates by creating a circle and then removing part of the circle, so fn and fa parameters apply to the entire circle not just the wedge.
    """    
    t = ops.circle(radius, fn=fn, fa=fa) if (fn and fa) else ops.circle(radius, fn=fn) if fn else ops.circle(radius, fa=fa) if fa else ops.circle(radius)
    if (angle >=0 and angle < 90):
        cutout = ops.polygon([[0,0],[radius*math.cos(math.radians(angle)), radius*math.sin(math.radians(angle))], [radius, radius], [-radius,radius],[-radius,-radius],[radius,-radius],[radius,0]])   
    elif (angle >=90 and angle < 180):
        cutout = ops.polygon([[0,0],[radius*math.cos(math.radians(angle)), radius*math.sin(math.radians(angle))], [-radius, radius], [-radius,-radius],[radius,-radius],[radius,0]])
    elif (angle >=180 and angle < 270):
        cutout = ops.polygon([[0,0],[radius*math.cos(math.radians(angle)), radius*math.sin(math.radians(angle))], [-radius,-radius],[radius,-radius],[radius,0]])
    elif (angle >=270 and angle < 360):
        cutout = ops.polygon([[0,0],[radius*math.cos(math.radians(angle)), radius*math.sin(math.radians(angle))], [radius,-radius],[radius,0]])
    else:
        return t  # full circle
    
    return t - cutout
    
def arc(radius, angle, thickness, fn=None, fa=None):
    """Returns a 2d arc. It operates by creating a circle and then removing part of the circle, so fn and fa parameters apply to the entire circle not just the wedge."""
    if thickness >= radius:
        return wedge(radius, angle)
    if thickness <= 0:
        return OP() # nothing
    else:
        outer = wedge(radius, angle)
        inner = wedge(radius-thickness, angle + 0.01)
        return outer - inner
    
    
def TestWedgeArc():
    a = ops.path_extrude(wedge(10,60, fn=16),[[0,0,0],[0,0,10]])
    b = ops.path_extrude(wedge(10,120, fn=30, fa=1),[[0,0,0],[0,0,10]]).translate([30,0,0])
    c = ops.path_extrude(wedge(10,210, fa=45),[[0,0,0],[0,0,10]]).translate([60,0,0])
    d = ops.path_extrude(wedge(10,300),[[0,0,0],[0,0,10]]).translate([90,0,0])
    e = ops.path_extrude(wedge(10,360),[[0,0,0],[0,0,10]]).translate([120,0,0])
    
    CH = 0
    width = 3
    f = ops.path_extrude(arc(10,60 - CH,width),[[0,0,0],[0,0,10]]).translate([0,30,0])
    g = ops.path_extrude(arc(10,120 - CH,width),[[0,0,0],[0,0,10]]).translate([30,30,0])
    h = ops.path_extrude(arc(10,210 - CH,width),[[0,0,0],[0,0,10]]).translate([60,30,0])
    i = ops.path_extrude(arc(10,300 - CH,width),[[0,0,0],[0,0,10]]).translate([90,30,0])
    j = ops.path_extrude(arc(10,360 - CH,width),[[0,0,0],[0,0,10]]).translate([120,30,0])
    
    return a | b | c | d | e   | f | g | h | i | j
```

gsohler, 2024-02-21:
the code works great and is definitely an enrichment to pythonscad.
However you  probably want to make it more fool-proof. dont relay on the import openscad as osc and make sure that math is imported.You could develop your library, publish it and i would link to it on the pythonscad homepage. what do you think ?

gandrewstone, 2024-04-01:
I don't think its sufficient functionality to warrant a separate library.

closed 2024-04-01 by gandrewstone