gsohler/openscad #21: Feedback on handles

gandrewstone, 2024-02-19:
First, they are a great idea and good implementation!  However, they really need to be a point vector (a point AND a vector) so that the handle can define both position and orientation.  And then we need a mode for your "align" function that equates the handle of one object with the handle of the other.  This will allow someone to use handles to reliably connect two object exactly where they are defined to connect, regardless of orientation or position.

gsohler, 2024-02-19:
Hi Andrew,
This is exactly my idea. The Handles are not coordinates only, but actually a 4x4 eigen matrix. which can store not only the position, but also the 3d direction of the target x, y and z axis in space. (yes, only position is displayed, but they are way more).
Did you try to rotate a handle before moving it to the target position ? You can even scale, mirror or skew handles like you can do
with solids. 
aligned_object = master.align(source.handle1, master.handle2)

output(source|aligned_object)
implements exactly what you are describing :)
let me know if you can recognize it ...

closed 2024-03-14 by gsohler

gandrewstone, 2024-03-15:
sorry for ghosting this, will try it out soon!

Captayne, 2024-03-17:
Sehr geehrter Herr Sohler

  wo bekomme ich denn die Libs her, die in den Beispielen verwendet
werden und wie installiert man sie, so dsas OpenScad sie findet? z.B.

from pylibfive import *
import pyqrcode

Bei mir gibts da von Hause aus eine Fehlermeldung:

arsing design (AST generation)...

ERROR: Traceback (most recent call last):

File "<string>", line 3, in <module>

ModuleNotFoundError: No module named 'pyqrcode'