gsohler/openscad #23: Apple Silicon Binaries

gsrunion, 2024-02-26:
Per the title, it appears the provided installer is for Intel silicon. When running the binary installed via that method...

% ./OpenSCAD
zsh: illegal hardware instruction  ./OpenSCAD

gsohler, 2024-02-26:
Hi,
Thank you for testing
Unfortunately I don't own a MAC myself and I got the chance to borrow one
some time back, which i real old one
and its apparently not compatible with recent models.
But I am Very interested in generating OpenSCAD images for recent macs.
All Volunteers are welcome, I can offer help giving hints for issues which
arise during compilation.

gsohler, 2024-06-06:
If somebody would grant me temporary user-only access to an apple-silicon mac, I would  try my best to compile pythonscad there and create a DMG file.

gsohler, 2024-07-07:
A Macbook having processor M1, M2, or M3 means its running on Apple Silicon, right ?

gsohler, 2024-07-23:
Today I have created a new Apple Disc image onto pythonscad.org
Almost all python features are there, let me know if its working

gsrunion, 2024-07-23:
Oooh I look forward to having a look.

Regards,
    Steve Runion

gsohler, 2024-08-10:
Latest Apple Silicon Binaries (2024-08-08) are reported working

closed 2024-08-10 by gsohler