gsohler/openscad #24: QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled

fzellini, 2024-04-28:
Hi, and thank you for your excellent work.
I have a problem with pythonscad:
On ubuntu22, 
I have downloaded openscad-2024.01.16.x86-64.tar.gz from https://pythonscad.org/download.php,
extracted and installed, but attempting to open a .scad file, there are  multiple messages like
"QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled"  and a segmentation fault

openscad 
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
Segmentation fault (core dumped)

gsohler, 2024-04-28:
Hi, Thank you for your interest in pythonscad.

I believe that qcb integration is related to using GL inside Qt.

your error shown appears to be a common issue, the internet is full of it.
you might look into this solution:

https://forums.autodesk.com/t5/installation-licensing/ubuntu-22-04-quot-cannot-create-platform-opengl-context-quot/td-p/11729065

This solution ultimately suggests to delete all libxcb stuff in the installation or rather replace all libxcb stuff with the files from the system instead.

Another option is to compile pythonscad from scratch, which is not difficult either.

As I did not change anything with the GL initialization in my fork, my guess is that 
the original software has the same issue in your system

deb https://download.opensuse.org/repositories/home:/t-paul/xUbuntu_22.04/ ./

Right now i have created a recent release package for linux, but i doubt it will make a difference.

Let me know if using the system libxcb files gives an improvement.

fzellini, 2024-04-29:
OK finally i have compiled all the stuffs in a docker container.
It seems to work well, the only issue is that "uni-get-dependencies.sh" does not install required dependencies nettle-dev and libjpeg-dev, so I added in the enclosed Dockerfile.
you can build with 
```
docker build . -t pythonscad
```
and run with
```
docker run --init  --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix --entrypoint /bin/bash  -e DISPLAY=$DISPLAY   -v ~/scad:/openscad  pythonscad
```
When inside docker, 
simply type "openscad --trust-python" to run the program
or, without bash:
```
docker run --init  --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix --entrypoint openscad  -e DISPLAY=$DISPLAY   -v ~/scad:/openscad  pythonscad
```

cat Dockerfile
```
FROM ubuntu:22.04

RUN apt update \
    && apt install -y --no-install-recommends \
    tar gzip git \
    x11-apps mesa-utils \
    apt-transport-https ca-certificates \
    nettle-dev libjpeg-dev

WORKDIR /work
RUN git clone https://github.com/gsohler/openscad.git
WORKDIR /work/openscad
RUN git checkout python
RUN git submodule update --init --recursive
RUN ./scripts/uni-get-dependencies.sh
RUN mkdir build
WORKDIR /work/openscad/build
RUN cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 ..
RUN make -j 8
RUN make install
RUN ldconfig
# optional
# RUN rm -rf /work
WORKDIR /openscad
```

Again, thank you for your excellent work. 

Fabrizio

gsohler, 2024-04-29:
Hi Fabrizio,

Thank you for your fast reply.

I don't know so much about docker, but I understand, that you compile and
run pythonscad in Docker , instead of Ubuntu now.
Even though the used commands look very exciting and promising , right now
i don't want to afford Docker due to my current disk space situation,
But ultimately I understand that pythonscad is running for you and maybe
you are exercising some of the shown examples.

Main discussion for pytonscad is at
 https://www.reddit.com/r/pythonscad/

So if you see issues, this is a great place to ask.
If you succeed and want to share your story,  feel free to share.

Thank you for pointing out the issue with nettle-dev and libjpeg-dev. Did
not notice these anymore, because these dependencies are installed in my
place for a long time.

closed 2024-10-08 by gsohler

mrdunk, 2024-11-28:
i had the same GLX issues so currently building from source.

a few more dependencies i needed:
System specs:
```
$ uname -a
Linux lapdancer 6.8.0-49-generic #49-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov  4 02:06:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux 
$ lsb_release -d
Description:	Ubuntu 24.04.1 LTS
```

Required dependencies:
`libtbb-dev qtmultimedia5-dev libqt5svg5-dev`

looking forward to getting this working.
i started a few days ago looking at FreeCAD to draw some geometry with repetitive elements that need modified with each invocation.
i suspect FreeCAD _could_ do what i want but the programmer in me was getting upset.
so i thought "OpenSCAD... i've always meant to try that out."
but a few hours later i was thinking "But how do i programmatically know what size a part is after i've applied boolean operations?"

And here i am.
This looks promising.
Thanks!

(build is at 12%....)

gsohler, 2024-11-28:
Hi Dunk!

Thank you for your interest in PythonSCAD !

... and sharing your experience with me!
I Just added the missing packets  into the  scripts/uni-get-depedencies.sh,
Thank you!

Also me is a very pragmatic programmer. Clicking  on something, especially
repeat the clicking make me insane!

So with OpenSCAD i never accepted the fact, that a variable cannot be
assigned once and you cannot retrieve coordinates from a mesh,

this is why i am here right now ... coding more functionality for PythonSCAD

PythonSCAD cannot directly retrieve the bounding box of an object, but
mesh() is your friend.
it delivers the coordinates of all vertices and you easily evaluate the
size of your object

.. hopefully build is more than 12% right now. i am happy to address any
error.

Look forward to hearing from you again

mrdunk, 2024-11-28:
thanks for the encouragement.
it built successfully and runs the Hello World (table) example.
i'll let you know how i get on.