gsohler/openscad #25: python stub file for openscad module

fzellini, 2024-05-01:
For python, there are many IDEs better than integrated openscad editor,
and, since openscad module  is implemented as native library, stub file is needed for syntax checking and autocompletion.

gsohler, 2024-05-01:
I like the idea of having python stub files(pyi). Everything which helps user experience will help to become popular.
(and its even some sort of quick cheat sheet showing all the functions and paramters)
Just now I am wondering, if i started to write the stub file, whats an easy editor to test them with ?

fzellini, 2024-05-02:
with pycharm you can  [create and check](https://www.jetbrains.com/help/pycharm/stubs.html#create-stub-local)  stub files

fzellini, 2024-05-02:
Seems very nice :-D 
Maybe a simple script that parse c++ code and generate python stub would help

```
# openscad.pyi
class PyOpenSCAD:
    def color (self, c:str, alpha: float, texture:str) -> PyOpenSCAD:
        ...
    def rotx (self, float) -> PyOpenSCAD:
        ...
    def translate (self, v:float | list[float]) -> PyOpenSCAD:
        ...

def cube( size:float | list[float], center: bool) -> PyOpenSCAD: ...
```

[image]

KlumpRasmus, 2024-05-02:
I am using VS Code. It would be great with type-hints and autocompletion.

gsohler, 2024-05-02:
already working on a stub file. hope to provide it today or tomorrow ...

gsohler, 2024-05-02:
i have checked 1st version of the stubs file

its in

https://github.com/gsohler/openscad/blob/python/libraries/python/openscad.pyi

not yet installed PyCharm due to size reasons.

anybody want to test the file  and send me improvements/corrections ?

fzellini, 2024-05-02:
Nice work, 
there are unresolved reference to PyOpenSCAD and PyLibFive, and  method parameters defined as "string" instead of "str".
relatively to PyOpenSCAD class, the methods of that object should be the ones listed [here](https://github.com/gsohler/openscad/blob/python/src/python/pyfunctions.cc#L3225) under OO_METHOD_ENTRY right ?

gsohler, 2024-05-02:
Hi fzellini. i think now python likes it.
Yes, you exactly found the place, where i define pythonscad methods. once the functions are settled, i will copy over to methods.

gsohler, 2024-05-08:
updated the stubs today. you are free to check now

closed 2024-09-02 by gsohler