gsohler/openscad #26: animation timings

petaflot, 2024-05-02:
disclaimer: I have been following this project for a while, but for various reasons have not been able to give it a try yet. forgive me if this "issue" has been fixed already.

animations are great to animate and test complex mechanical systems such as geared reductors and many other things. in such cases, one really wants the animation to loop over cleanly. ini this aspect, having the timing variable `$t` going from zero to one is very unpractical, because it should be obvious by now that we have a not much else than a rotating vector: the timing variable should loop over the [0:2pi] range (or [-pi:pi]).

because of backwards-compatibility issues this is not something I would request in the "legacy" openscad, however, since there are probably not too many python-openscad models out there, I believe this should be done ASAP.

petaflot, 2024-05-02:
> Other option could be that pythonscad would automatically define variable "phi" for the python script which would go from 0 to 2pi.

this would be good, much better than having to specify it manually: the main reason is that I see some UI improvements related to this, and having to make the widgets react to a non-standard variable seems like it will be more work and more likely to cause confusion.

+1 for automatic definition of `phi`

gsohler, 2024-05-02:
[image]

it allows for really nice code

petaflot, 2024-05-02:
I couldn't agree more :-)

closed 2024-09-02 by gsohler