gsohler/openscad #28: down() is broken.

willywong, 2024-08-18:
**Describe the bug**

One of the translation methods, down(), is broken.

`AttributeError: PyOpenSCAD object has no attribute 'down'`

**To Reproduce**
Steps to reproduce the behavior:

Sample code:

```py
from openscad import *
def test2():
    x = cube(3).color('green')
    y = cube(2).color('red').down(10)
    return y

z = test2()
output(z)
```

It will die with `AttributeError: PyOpenSCAD object has no attribute 'down'`

**Expected behavior**

Down works.

**Code reproducing the issue**

See above.

**Screenshots**

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 (x64)
 - System: Intel(R) Pentium(R) CPU G4560 @ 3.50GHz   3.50 GHz
 - OpenSCAD Version PythonSCAD_nolibfive-2024.08.13-x86-64-Installer.exe

gsohler, 2024-08-19:
Yes, confirmed, one single Line got deleted in the source. Can fix in 2 Weeks. Temporary fix could up(-value)
Thx!

gsohler, 2024-09-02:
The issue is fixed. down() will be available in next release again

gsohler, 2024-09-03:
new version 09-03 is online

closed 2024-09-07 by gsohler