gsohler/openscad #29: projection() does not seem to be working

easytarget, 2024-08-19:
I'm trying to do a 2d projection of a 3d object, equivalent to `projection()` in SCAD
```python
from openscad import *
c = cube([10,10,10],center=True)
p = projection(c)
output(p)
```
Should produce a centered square (2d object), but seems to produce an empty 2d object instead. I've tried this several ways and cannot seem to make it work. 

2024-08-10.AppImage on Fedora40.

gsohler, 2024-08-19:
Hi,
Thank you for the Bug Report. This should be fixes by 2024-08-13. Version.
Intermediate fix should be disabling Manofold temporarily. Let me know...

easytarget, 2024-08-19:
Thank-you. Yes, disabling Manifold in edit->preferences->features allows this to work. Sorry for the interruption ;-)

closed 2024-08-19 by easytarget