gsohler/openscad #32: Issue with variable scope for a global variable?

WillAdams, 2024-09-02:
Given a Python file:

    #!/usr/bin/env python
    
    def psetupstock(stocklength, stockwidth, stockthickness, zeroheight, stockorigin):
        global mpx
        mpx = float(0)
        global mpy
        mpy = float(0)
        global mpz
        mpz = float(0)
        global tpz
        tpz = float(0)
        global currenttool
        currenttool = 102
    
    def psettool(tn):
        global currenttool
        currenttool = tn
    
    def pcurrent_tool():
        global currenttool
        return currenttool

which one makes use of in a .scad file containing:

    use <gcptest.py>;
    
    /* [CAM] */
    small_square_tool_no = 122; // [0:0,122:122,112:112,102:102]
    
    psetupstock(1, 2, 3, "Top", "Center");
    
    echo(str(pcurrent_tool()));
    
    psettool(small_square_tool_no);
    
    echo(str(pcurrent_tool()));

The first invocation of pcurrent_tool() results in:

[WARNING: Ignoring unknown function 'pcurrent_tool' in file gcptest.scad, line 8](https://github.com/gsohler/openscad/issues/8,C:/Users/willa/OneDrive/Desktop/gcptest.scad)
ECHO: "undef"

Is this a bug? Or expected behaviour? 

If it is expected, could someone point me to a discussion of variable scope in Python which explains this point which I am missing?

WillAdams, 2024-09-02:
When all this was copied into a single Python file:

[image]

it ran as expected:

>Python 3.11.5 (tags/v3.11.5:cce6ba9, Aug 24 2023, 14:38:34) [MSC v.1936 64 bit (AMD64)] on win32
>Type "help", "copyright", "credits" or "license()" for more information.
>
>= RESTART: C:/Users/willa/OneDrive/Desktop/pythontest.py
>102
>122

gsohler, 2024-09-02:
No, its much worse,
pythonscad does simple not yet support returning integers from python back to scad yet, even though it was simple.
Just lists and strings are working yet.
if you set currenttool to "102" , it works as expected

You might want to keep this open until the "issue" is solved.

WillAdams, 2024-09-02:
Isn't the 122 in the above Pythonscad returning a value?

gsohler, 2024-09-02:
yes, psettool returns the value 102, but its an integer, and returning integers from python to scad is just not yet supported.
(only floats, lists and strings)

just now coding the line.

gsohler, 2024-09-02:
[image]
working locally already

WillAdams, 2024-09-02:
That's great to see. Thanks! Looking forward to the next version!

WillAdams, 2024-09-02:
Is this command:

    echo(str(ptool_diameter(pcurrent_tool(), stockthickness)));

resulting in:

[WARNING: Ignoring unknown function 'ptool_diameter' in file gcodepreview_unittests.scad, line 64](https://github.com/gsohler/openscad/issues/64,C:/Users/willa/OneDrive/Desktop/gcodepreview_unittests.scad)
ECHO: "undef"

this same issue? Already fixed?

gsohler, 2024-09-03:
new version 09-03 is online

WillAdams, 2024-09-03:
Thanks! I've begun testing, it's been working perfectly, and it looks like I should have my Python unit-tests done presently, then I'll get into the OpenSCAD ones, then get back to "real" developing.

closed 2024-10-08 by gsohler