gsohler/openscad #34: align() resets shape scaling.

willywong, 2024-09-05:
**Describe the bug**
align() resets shape scaling.

**To Reproduce**
Run this in PythonSCAD:

```py
from openscad import *

# mentos looking shape.
mentos = sphere(10).scale([1, 2, 2])

# box moved off center for align use.
box = cube(5).right(50)

# This moves the mentos, but also reset its shape back to a normal sphere.
why_is_this_a_sphere = mentos.align(box.origin, mentos.origin)

output([why_is_this_a_sphere, mentos])
```

**Expected behavior**

The mentos looking shape is moved, but retains its shape?

**Code reproducing the issue**

See above.

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel
 - OpenSCAD Version 2024.09.03. Downloaded Windows installer from pythonscad.org

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

```
OpenSCAD Version: 2024.09.03 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2024.09.03 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/PythonSCAD/bin
Documents Path: C:\Users\wiw\Documents
User Documents Path: C:\Users\wiw\Documents
Resource Path: C:/Program Files/PythonSCAD
User Library Path: C:/Users/wiw/Documents/OpenSCAD/libraries
User Config Path: C:\Users\wiw\AppData\Local/OpenSCAD
[... 294 lines cut ...]
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

```

gsohler, 2024-09-05:
When you scale the sphere,, you do not only scale the geometry, but also its handles.(origin)
Later when you use mentos.origin as reference you literally "undo"  the skew by using scaled handles.
You can fix your issue  by not specifying mentos.origin in the align command.

willywong, 2024-09-05:
>  You can fix your issue by not specifying mentos.origin in the align command.

Wonderful. `mentos.align(box.origin)` works as expected.

I did not realize the second parameter was optional! All the examples I encountered had it supplied.

I can make an update to https://www.reddit.com/r/pythonscad/wiki/index#wiki_align.

closed 2024-09-05 by willywong

gusohler, 2024-09-06:
Thank you for confirming it fixed. Your OK Info ob Wiki IS Not correct.
Will try to fix IT.