gsohler/openscad #37: 8.10 AppImage does not contain Python requirements

gatesphere, 2024-09-17:
**Describe the bug**
The 8.10 AppImage does not contain the required Python libraries to run standalone.

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to run the 8.10 AppImage on a Linux machine without Python 3.11 installed, and render a test .py file
2. Observe the console output

**Expected behavior**
An AppImage should contain all requirements for an application to run standalone.

**Code reproducing the issue**
N/A

**Screenshots**
N/A

**Environment and Version info (please complete the following information):**
N/A

**Library & Graphics card information**
N/A

**Additional context**
This may be a good thing -- a bundled Python 3.11 inside an AppImage would have a hard (impossible?) time using any additionally installed pip packages that a user may install separately.

gsohler, 2024-09-17:
Thank you for reporting the bug and your analysis.
The AppImage did run in my computer as it got correct python from somewhere else. 
I am happy that others tell me, that its not self contained.
The AppImage was created with "linuxdeploy", need to find a way  to include python libraries in addition.
If you have some experience, i'd be happy to know.

gsohler, 2024-09-17:
AppImages can read files from the users local computer(e.g. when  do file->open->Python/SCAD Design)
I hope that same also applies for python libraries installed in the host computer.

rp10007, 2024-09-18:
Another observation: I've got   3.11installed, but the  console tells me: 

ERROR: Python not found. Is it installed ?\

FWIW, another big python-based appimage,  mu-editor,  apparently  relies on its own venv and pip setup to install libraries in a special  space that doesn't talk to system python installations. I don't know whether that's something you want to do,  or to look at for clues.

gsohler, 2024-09-18:
FYI, i created yet another test version of the App-Image with an explicit
python-plugin and
the result is a ridiculous huge (250MB) appimage
its in http://pythonscad.org/OpenSCAD-x86_64.AppImage and of course it
works for me.
Not sure it works for others

gsohler, 2024-09-18:
> OK, this is weird.
>
> First.py is a cut-and-paste of the table example from the web page:
>
> def foot(x,y):
> 	c=cube([1,1,10])
> 	return translate(c,[x,y,0])
>
> def plate():
> 	return cube([11,11,1])
>
> parts=[]
> parts.append(translate(plate(),[0,0,10]))
>
> for y in [0,10]:
> 	for x in [0,10]:
> 		parts.append(foot(x,y))
>
> output(parts)
>
> Console says:
>
> Loaded design '/home/paul/Documents/scad/pythonscad/First.py'.
> ERROR: Python not found. Is it installed ?
> Execution aborted
>
> I wonder whether there's something in the Appdir setup or the paths that's
> not getting rewritten properly, so that the paths still work for you but
> not anyone else? (Just a cursory look at linuxdeploy, so I could be
> completely off base...)

hi Paul

let me share the notes, how i create the AppImage

cd build
cmake  -DCMAKE_INSTALL_PREFIX=/usr -DEXPERIMENTAL=1 -DENABLE_PYTHON=1
-DENABLE_LIBFIVE=1 ..
make -j3
rm -rf ../AppDir
make install DESTDIR=../AppDir
cd ..
export EXTRA_QT_PLUGINS=svg
/home/gsohler/git/linuxdeploy-plugin-python/linuxdeploy-plugin-python.sh
--appdir AppDir
export PATH=/appimage/usr/local/bin:$PATH
linuxdeploy --plugin qt --output appimage --appdir AppDir

linuxdeploy-plugin-python is from
 git clone https://github.com/niess/linuxdeploy-plugin-python.git

i realized, that the script run quite straight forward and even compiled a
complete python

just my 2 cents (for the moment)
need to approach next day now  :)

hradec, 2024-10-08:
I just tried to run the latest appimage from today on a debian 12 distro, running python 3.11 and got this error on the shell window, when I loaded a .py script: 
```
Python Code globally trusted
Python path configuration:
  PYTHONHOME = '/usr'
  PYTHONPATH = '/tmp/.mount_PythonShYnLP/usr/bin/../libraries/python/:.'
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = '/usr/lib64/python3.11'
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.platlibdir = 'lib64'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/tmp/.mount_PythonShYnLP/usr/libraries/python',
    '/tmp',
    '/usr/lib64/python311.zip',
    '/usr/lib64/python3.11',
    '/usr/lib64/python3.11/lib-dynload',
  ]
XXX readobject called with exception set
```
Debian doesn't have `/usr/lib64/python3.11`, but it does have `/usr/lib/python3.11`
I also noticed the `PYTHONPATH = '/tmp/.mount_PythonShYnLP/usr/bin/../libraries/python/:.'`, and I went to check in the appimage mounted folder, there's no /tmp/.mount_PythonShYnLP/usr//libraries/python/:
```
rhradec@server /usr/lib64 $ ls -l /tmp/.mount_PythonShYnLP/usr/
total 0
drwxr-xr-x.  2 root root 0 Aug 10 12:16 bin
drwxr-xr-x.  2 root root 0 Aug 10 12:16 lib
drwxr-xr-x. 12 root root 0 Aug 10 12:16 plugins
drwxr-xr-x.  8 root root 0 Aug 10 12:12 share

```
I looked for this `libraries` folder, and actually found it in `/tmp/.mount_PythonShYnLP/usr/share/openscad/libraries/`: 
```
rhradec@server /usr/lib64 $ ls -l /tmp/.mount_PythonShYnLP/usr/share/openscad/libraries/
total 0
drwxr-xr-x. 3 root root 0 Aug 10 12:12 MCAD
drwxr-xr-x. 2 root root 0 Aug 10 12:12 python
```
So it seems PYTHONPATH is defined to the wrong path in the appimage as well. 

I also think the appimage should have python and all the needed python modules and libraries in it, so it can run out of the box without relying in the distro python version and setup.

gsohler, 2024-10-08:
Hi Hradic, thank you for you interest in PythonSCAD and improving the AppImage.
I  completely agree, that python libs should be included in the AppImage.
This is, why found a plugin to linuxdeploy, which claims beeing able to add Python inside to the AppImage
I found it at //github.com/niess/linuxdeploy-plugin-python.git and i was even able to configure for my favorite python version.

It appeared to work, and the result is an AppImage which is 250 MB in size.
If you are interested to test this ridiculous big file, you can download it at

https://pythonscad.org/OpenSCAD-x86_64.AppImage

which I just created few minutes back to tell me, if it works for you. (for me it works) 

If you interested we can sync and i can show you, how I create the AppImage, so we can both work on your situation. 
Let me know

rp10007, 2024-10-10:
When  I try the new one, I get the  following messages in the terminal:
```
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Fatal Python error: PyGILState_Release: thread state 0x2588c00 must be current when releasing
Python runtime state: unknown

Aborted (core dumped)
```
If I select "new" from  my opening popup  window,  I get to type in  the edit window, and the program only crashes when I try to render with F5. If I select a recent openscad or python file, it crashes immediately.  (Also, when I  select "new" there's a small window in the center of the screen that's titled "openscad" and has all-black content.)

This is on Ubuntu 22.04.5 with Gnome and Wayland.

I think there's something in environments where the big blob works that's still not being captured by the appimage, alas.

gsohler, 2024-12-18:
today I have redone the PythonSCAD AppImage with latest additions.
Its available at https://pythonscad.org/download.php as PythonSCAD-2024-12-18.AppImage.
To test if Python was completely embedded, i moved away my system python libraries for a moment and the AppImage was still executing Python.
Also the PyGILState was fixed quite some time ago.

rp10007, 2024-12-19:
Alas, still asking me where Python is. It displays openscad code just fine, but a python file gives me 

ERROR: Python not found. Is it installed ?

This is what the code says when it runs in a terminal:

`qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = '/tmp/.mount_Pythonv7ay7c/usr/bin/../libraries/python:/home/xxxxxx/.local/share/OpenSCAD/libraries:.'
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = '/usr/lib64/python3.12'
  sys._base_executable = '/usr/local/bin/python3'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.platlibdir = 'lib64'
  sys.executable = '/usr/local/bin/python3'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/tmp/.mount_Pythonv7ay7c/usr/libraries/python',
    '/home/xxxxxl/.local/share/OpenSCAD/libraries',
    '/home/xxxxx/Documents/scad/pythonscad',
    '/usr/lib64/python312.zip',
    '/usr/lib64/python3.12',
    '/usr/lib64/python3.12/lib-dynload',
  ]
`
Meanwhile, some scouting shows 

>ls /tmp/.mount_PythonIYEBFe/usr/bin/
openscad  qt.conf

>ls /tmp/.mount_PythonIYEBFe/usr/bin/openscad 
/tmp/.mount_PythonIYEBFe/usr/bin/openscad

>ls /tmp/.mount_PythonIYEBFe/usr/
bin  lib  plugins  share

/usr/lib64:
ld-linux-x86-64.so.2

(with the python items in /usr/lib)

Oh, and anything I've tried to put in PYTHONHOME -- /usr/ ,  /usr/bin/ , /usr/local/ , -- hasn't changed things. It removes the error messages about not being able to find platform-independent or platform-dependent libraries, but the in-app message about not being able to find python is still there.

I hope any of this helps to clarify things.

gsohler, 2024-12-19:
Hi Paul,
Thank you very much for your extensive tests.

The command line logs yield quite a lot of information when you are able to understand it.
It turns out when I run my  AppImage here, it reads its std lib from

/usr/lib64/python3.12/
And of course i have this directory and everything works fine her locally
(found out with strace)

And as soon as i move this dir away, I also get error "is it installed ?" - same as you are seeing.

So  my task was to add yet another library path for python where it was find its std lib,
this is /tmp/.mount_Pythonv7ay7c/usr/bin/../lib/python3.12/
in your case

The resulting version even works, when i move my python shared libs and python std lib files awy
So i really hope, it works for you too, this time

Latest appimage Version is from 2024-12-19

rp10007, 2024-12-20:
Thank you! No extensive testing yet, but it renders the simple python table example (as soon as I remembered to include the "from openscad import *" line.) Looks good.

It would be nice to have an opening script that would find the local version of python and mash that into the appropriate environment and python sys variables before launching openscad, so you could slim the Appimage back down, but you've already been doing so much. And this will definitely do for the foreseeable future. And thanks for the fast response.

gusohler, 2024-12-20:
Thank you for your patience.
Not sure if we can skip the python part from the app image.
AppImage is supposed to work under all circumstances and if the user does
not have *any* python installed, this would be a clear fail.
please share your experience about PythonSCAD with us

weerasuriya, 2025-03-17:
I appear to have the same problem on the latest version of the AppImage. I note also that the AppImage seems to be looking for Python3.11, but Python3.12 is bundled?

Command line output:

```bash
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
++
--
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = '/tmp/.mount_PythoncnQtLb/usr/bin/../libraries/python:/tmp/.mount_PythoncnQtLb/usr/bin/../lib/python3.11:/home/chathika/.local/share/OpenSCAD/libraries:.'
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = '/usr/lib64/python3.11'
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.platlibdir = 'lib64'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/tmp/.mount_PythoncnQtLb/usr/libraries/python',
    '/tmp/.mount_PythoncnQtLb/usr/lib/python3.11',
    '/home/chathika/.local/share/OpenSCAD/libraries',
    '/tmp',
    '/usr/lib64/python311.zip',
    '/usr/lib64/python3.11',
    '/usr/lib64/python3.11/lib-dynload',
  ]
XXX readobject called with exception set
```

My system Python version is 3.13, but I have tried launching the AppImage in 3.11/3.12 virtualenvs, creating various virtualenvs and setting PYTHONPATH/HOME to them etc, to no avail...

gsohler, 2025-03-18:
H weerasuriya, this is great info.
I will carefully review my AppImage process

gsohler, 2025-03-18:
I have now recreated the AppImage just to get a "latest" version.
Now I checked, that consistently contains 3.12.0 in the core and in the bundled ptython.
Can you check on PythonSCAD-2025-03-18.AppImage ?

jankrahl, 2025-04-02:
I still get the error with PythonSCAD-2025-03-18.AppImage.

I am not an expert at all with AppImages but when I run it from the terminal I get the following output:

```Could not find platform independent libraries <prefix>
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'openscad-python'
  isolated = 0
  environment = 1
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = '/usr/lib64/python3.12'
  sys._base_executable = '/tmp/.mount_Pythonxpff20/usr/bin/openscad-python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/tmp/.mount_Pythonxpff20/usr'
  sys.platlibdir = 'lib64'
  sys.executable = '/tmp/.mount_Pythonxpff20/usr/bin/openscad-python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/tmp/.mount_Pythonxpff20/usr'
  sys.path = [
    '/usr/lib64/python312.zip',
    '/usr/lib64/python3.12',
    '/tmp/.mount_Pythonxpff20/usr/lib64/python3.12/lib-dynload',
  ]
```

I notice that `PYTHONHOME` and `PYTHONPATH` are not set and that `sys.path` and `stdlib dir` do **not** point into the AppImage (i.e. `tmp/.mount_Pythonxpff20`).
I supect that one of these things could be the reason.

jankrahl, 2025-04-02:
`ln -s /usr/lib/python3.12 /usr/lib64/python3.12` indeed fixed it for me, so it seems that even though the AppImage has the correct python version bundled but it doesn't use it.

gsohler, 2025-04-02:
Hi Jan,

Thanks for your great investiagtion. Your conclusions were absolutely correct.
When merging the python work from  upstream openscad, we were compiling
an 1a grade sys.path, but its no useful when not setting it.

https://imgur.com/a/ibF92FB

Just now I created a new AppImage, run it and checking that the sys.path actually also sets into the AppImage's python installation

https://imgur.com/a/BO52nN5

I  have just uploaded that AppImage to pythonscad.org in case you  are interested.

Thx again for your investigations.

closed 2025-04-20 by gsohler