gsohler/openscad #38: Using OpenSCAD libraries from Python code

adob, 2024-09-30:
When modeling using Python, I would like to be able to make use the BOSL library, which is written in OpenSCAD.

For example, instead of using the OpenSCAD cuboid primitive, I would like to use the cuboid module provided by BOSL. There doesn't appear to be a way of importing OpenSCAD code into Python to enable this.

gsohler, 2024-09-30:
Hi,
Thank you for your interest!
We are already working on this ...

https://www.reddit.com/r/pythonscad/comments/1fidygn/reusing_your_existing_scad_libraries_in_pythonscad/

Unfortunately its not that easy  without support from main openscad devs(they have other prios right now)
any help is appreciated to make this goal come true earlier

gsohler, 2024-12-16:
Finally managed to use BOSL2 in PythonSCAD

https://www.reddit.com/r/OpenPythonSCAD/comments/1hf2bum/using_existing_scad_libraries_in_pythonscad/

checkout 2024-12-15 version to use them

willywong, 2025-01-22:
I can confirm osinclude() works on version 2025.01.02 downloaded from pythonscad.org (Windows version).

@adob would you like to test and confirm on your end?