gsohler/openscad #39: Call OpenSCAD functionality in Python file from Python file

WillAdams, 2024-10-01:
Duplicate of: https://old.reddit.com/r/pythonscad/comments/1fqzw2s/best_practices_for_global_variables_in/

Given a Python file, pythontest.py which contains:

    from openscad import *
    
    def popengcodefile(fn):
        global f
        f = open(fn, "w")
    
    def writeln(*arguments):
        global f
        line_to_write = ""
        for element in arguments:
            line_to_write += element
        f.write(line_to_write)
        f.write("\n")
    
    def psetupstock(stocklength, stockwidth, stockthickness, zeroheight, stockorigin):
        global Base_filename
        global generategcode
        global stock
        stock = os.cube([stocklength, stockwidth, stockthickness])
        if generategcode == True:
            writeln("(Design File: " + Base_filename + ")")

and a test.py file which calls it as:

    from openscad import *
    from pythontest import *
    
    # [Export] */
    Base_filename = "export" 
    # [Export] */
    generategcode = True
    
    global filename_gcode
    filename_gcode = Base_filename + ".nc"
    
    popengcodefile(filename_gcode)
    #pythontest.popengcodefile("export_set.nc")
    
    psetupstock(1,2,3,"Top","Center")
    #pythontest.psetupstock(1,2,3,"Top","Center")
    
    # Create a cube and a cylinder that overlap
    cy = cylinder(15, center = True)
    
    # Substract the cylinder from the cube
    diff = stock.difference(cy)
    #diff = pythontest.stock.difference(cy)
    
    # Display the result
    output(diff)

the result is:

>    ERROR: Traceback (most recent call last):
>      File "<string>", line 15, in <module>
>      File "C:\Users\willa\AppData\Local\Programs\Python\Python311\Lib\pythontest.py", line 17, in psetupstock
>        global generategcode
>                ^^^^
>    NameError: name 'cube' is not defined

What arrangement of commands and files would allow one to:

 - define 3D modeling elements in a .py
 - import that .py file and call defined commands from it and get the 3D models instantiated

?

gsohler, 2024-10-01:
Fortunately this is an easy one.
in pythontest.py you included openscad plain into context
so you have to write:
stock=cube(xxx)

or you can import as
import openscad as os

hope this helps

WillAdams, 2024-10-01:
Changing that to:

        stock = cube(stocklength, stockwidth, stockthickness)

I still get:

>ERROR: Traceback (most recent call last):
>  File "<string>", line 15, in <module>
>  File "C:\Users\willa\AppData\Local\Programs\Python\Python311\Lib\pythontest.py", line 17, in psetupstock
>    global generategcode
>            ^^^^
>NameError: name 'cube' is not defined

Is there a sample/template of a pair of .py file where one has global variables, and the other does branching/3D modeling based on those variables when called from the first file?

WillAdams, 2024-10-01:
I get:

>ERROR: Traceback (most recent call last):
>  File "<string>", line 4, in <module>
>NameError: name 'gcodewriter' is not defined

gsohler, 2024-10-02:
Glad that you found a temporary solution, but it would be much nicer when you could leave  the gcodewriter library stuff,  where it is supposed to stay. I has to work!
when you import pythontest to *, also gcodewriter must arrive in the current namespace.

I think you are experiencing some caching issue.
* Can you try to edit the files and start openscad after you have finished and saved the files ?
* Can you look for *.pyc files and remove them before restarting openscad ?
* Can you  rename pythontest.py to another filename and use this one instead before restarting openscad ?

I really hope,that one of these must help. Depending on the outcome we can proceed from there.

WillAdams, 2024-10-02:
Good point. Found a cache file in:

    C:\Users\willa\AppData\Local\Programs\Python\Python311\Lib\__pycache__

for pythontest.py and when I deleted that and re-opened test.py it worked as expected:

[image]

So should be good to go! Thanks!

closed 2024-10-02 by WillAdams

gsohler, 2024-10-02:
Glad that we aware of cache files now.
I fear that cache thing has driven you crazy for a long time already without knowing

WillAdams, 2024-10-02:
Specifically added a section on this:

https://old.reddit.com/r/pythonscad/wiki/index#wiki_troubleshooting_and_cache_files

I am going to add a line to my "install.bat" file which deletes any cache files.