gsohler/openscad #41: Build fails on Linux

KlumpRasmus, 2024-10-15:
**Describe the bug**
Failure to build OpenSCAD Python branch on Linux platform.
Build fails with these errors:

```
-- Detecting CXX compile features - done
CMake Warning at CMakeLists.txt:116 (find_package):
  By not providing "FindSanitizers.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Sanitizers", but CMake did not find one.

  Could not find a package configuration file provided by "Sanitizers" with
  any of the following names:

    SanitizersConfig.cmake
    sanitizers-config.cmake

  Add the installation prefix of "Sanitizers" to CMAKE_PREFIX_PATH or set
  "Sanitizers_DIR" to a directory containing one of the above files.  If
  "Sanitizers" provides a separate development package or SDK, be sure it has
  been installed.

-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
CMake Error at submodules/CMakeLists.txt:47 (add_subdirectory):
  The source directory

    /home/poul/Temp/openscad/submodules/mimalloc

  does not contain a CMakeLists.txt file.

...
-- Found Nettle: /usr/include
CMake Error at CMakeLists.txt:456 (add_subdirectory):
  The source directory

    /home/poul/Temp/openscad/submodules/libfive

  does not contain a CMakeLists.txt file.

-- Python enabled, using Nettle 

...

-- Found Python3: /home/poul/.pyenv/shims/python3 (found suitable version "3.12.7", minimum required is "3.4") found components: Interpreter 
CMake Error at CMakeLists.txt:1131 (add_subdirectory):
  The source directory

    /home/poul/Temp/openscad/submodules/manifold

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:1137 (add_sanitizers):
  Unknown CMake command "add_sanitizers".

-- Configuring incomplete, errors occurred!

```

**To Reproduce**
Steps to reproduce the behavior:
Follow the build steps from https://pythonscad.org/

**Expected behavior**
Build completes producing an OpenSCAD binary.

**Code reproducing the issue**
NA

**Screenshots**
NA

**Environment and Version info (please complete the following information):**
Ubuntu 22.04, Intel I5

**Library & Graphics card information**
NA

**Additional context**
NA

gsohler, 2024-10-15:
can you check if there are files in
submodules/manifold 

if not, can you make sure that command
git submodule update --init --recursive
worked without error ?

KlumpRasmus, 2024-10-15:
No files in submodules/manifold. If I run `git submodule update --init --recursive` I get this:

fatal: remote error: upload-pack: not our ref 48a28df44ab9bd20d1fe2b2b3bbf29189e45aeb5
fatal: Fetched in submodule path 'submodules/libfive', but it did not contain 48a28df44ab9bd20d1fe2b2b3bbf29189e45aeb5. Direct fetching of that commit failed.

gsohler, 2024-10-15:
Thank you for pointing out.
I was accidently commiting the libfive repo (locally) and documenting the new hash whereas the original repo has never seen that.
i fixed now ;)
if you dont want to use libfive, you can disable it in the command line settings

KlumpRasmus, 2024-10-15:
Great, thank you :-) I can now build if I set PYTHON_VERSION with `-DPYTHON_VERSION=3.12`

gsohler, 2024-10-15:
glad to learn, PYTHON_VERSION is rather new feature, Used it to debug GIL locking for Python Threads

KlumpRasmus, 2024-10-15:
Hmm... that is I can build it, but I get an error when I open a python design:

```
libs is 
Fatal Python error: init_threadstate: thread state already initialized
Python runtime state: initialized

Aborted (core dumped)
```

I will try with Python 3.11 instead.

gsohler, 2024-10-15:
with lib3.11 python thread is not critical,
with lib3.12 it is .. apparently still not correctly implemented.
Its everything about python_lock and python_unlock function in pyopenscad.cc

KlumpRasmus, 2024-10-17:
If I disable python_lock/unlock altogether the application runs fine (Python 3.12) but it might mess with the newly implemented threading in the rest of the code?

gsohler, 2024-10-17:
openscad/pythonscad uses threads only to keep the GUI alive while the engine is calculating. Even though there are threads, the threads executing python do not ever overlap in time.
happy to realize, that pythonscad  works in your place without the locks.
For python3.12 it only works !with! locks.  very strange.

closed 2024-10-20 by gsohler