gsohler/openscad #43: Edit | Uncomment removes first character of commented line

WillAdams, 2024-10-20:
Given a set of lines in a file which have been commented out by the command Edit | Comment:

    #fn=20
    #def knot(t):
    #  return [ sin(t)+2.0*sin(2.0*t), cos(t)-2.0*cos(2.0*t),sin(3.0*t)]
    #n=150
    #pts=[]
    #for i in range(n):
    #    pts.append(knot(2*3.1415926*i/n))
    #path_extrude(square(0.5),pts,twist=-120,closed=True).output()

Selecting them and using Edit | Uncomment results in:

    n=20
    ef knot(t):
       return [ sin(t)+2.0*sin(2.0*t), cos(t)-2.0*cos(2.0*t),sin(3.0*t)]
    =150
    ts=[]
    or i in range(n):
       pts.append(knot(2*3.1415926*i/n))
    ath_extrude(square(0.5),pts,twist=-120,closed=True).output()

gsohler, 2024-10-20:
This is a good one. Appears, that the uncomment feature is very designed for SCAD language.
I am sure, we can improve here

gsohler, 2024-10-20:
was fixed by this commit
https://github.com/openscad/openscad/commit/e47913f9ef7c23f6fdccfff5ad90bc910449aed4
Thank you for finding!

gsohler, 2024-11-04:
working for you ?

WillAdams, 2024-11-04:
Working as expected for Python files in the new version as of 11/04. 

Thanks!

closed 2024-11-04 by WillAdams