gsohler/openscad #45: offset() on a 3d shape can crash

willywong, 2024-10-26:
**Describe the bug**
offset() on a 3d shape can crash.

Oddly enough, I am only able to reproduce this if (1) The 3d shape dimensions are large, and (2) the offset r is small.

**To Reproduce**

This code crashes it.

```py
from openscad import *

s = sphere(40)
ss = s.offset(r=-0.4)
show(ss)    # Comment this out, and it won't crash. Show(s) also will not crash.
```

**Expected behavior**
Not crash.

**Code reproducing the issue**
See above.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.09.03 
 -
**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

```
OpenSCAD Version: 2024.09.03 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2024.09.03 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg*, fast-csg-safer, fast-csg-debug, manifold*, roof*, input-driver-dbus, lazy-union*, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/PythonSCAD/bin
Documents Path: C:\Users\wiw\Documents
User Documents Path: C:\Users\wiw\Documents
Resource Path: C:/Program Files/PythonSCAD
User Library Path: C:/Users/wiw/Documents/OpenSCAD/libraries
User Config Path: C:\Users\wiw\AppData\Local/OpenSCAD
Backup Path: C:/Users/wiw/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/wiw/Documents/OpenSCAD/libraries
C:/Program Files/PythonSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/wiw/AppData/Local/Microsoft/Windows/Fonts
[... 284 lines cut ...]
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

```

willywong, 2024-10-26:
Interestingly enough, the openscad doc indicates offset() should only operate off of 2d shapes: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#offset

However, if I replace the above code from sphere to `cylinder(r=40, h=30)`, offset actually work as expected.

gsohler, 2024-10-26:
the bug was fixed recently. please compile latest pythonscad from source and it will work.
yeahh, pythonscad as been working on 3d offset ....

willywong, 2024-10-27:
To compile for windows, are the instructions in the README at https://github.com/gsohler/openscad?tab=readme-ov-file#building-for-windows accurate? They look like instructions for vanilla openscad and not pythonscad.

Are there additional dependencies/steps to compile pythonscad from source?

gsohler, 2024-10-27:
These instructions are quite recent.
Only missing information is that that you need to compile in branch "python"

But I am sorry, as I was not aware that you are using PythonSCAD in Windows
where compilation takes a lot of time.
If you like you dont need to take this step as i am releasing a new
version, soon.

willywong, 2024-10-28:
Thanks, that works. I can wait for the new release. Thank you!

gsohler, 2024-11-04:
new windows version 11-04 is available. can you check if offset works for you there ?

willywong, 2024-11-08:
Will try it out this weekend 👍

willywong, 2024-11-09:
Hey @gsohler , I tried 11-04 for Windows. It is still crashing.

gsohler, 2024-12-03:
Hi @willywong , finally got some time in looking after offset() again
some object faces combines accidently even when their faces were not exactly planar.
Your sphere example and many others do not crash anymore.
Now 2023-12-02 is online.

willywong, 2024-12-07:
Brilliant. I tested on 12.04, and it no longer crashes.

Thank you!

closed 2024-12-07 by willywong