gsohler/openscad #46: Shapes produced by wrap() cannot intersect or union with other objects without being filled.

willywong, 2024-11-16:
**Describe the bug**
Shapes produced by wrap() cannot intersect or union with other objects without being filled.

**To Reproduce**
Steps to reproduce the behavior:

Run the following code.

```py
from openscad import *

import math

R = 10

# Wide board standing on X-axis.
sheet = cube([2 * R * math.pi, 1, 20])
circular_shell = sheet.wrap(r=R)

plate = circle(r=2 * R).linear_extrude(1)

# Expect: hollow tube on top of a plate.
# Actual: The hollow tube gets filled.
show([circular_shell, plate])
```

**Expected behavior**

Expect: hollow tube on top of a plate.
Actual: The hollow tube gets filled.

**Code reproducing the issue**
See above.

**Screenshots**

Expected, and I can see this when F5.
[image]

When F6, the hollow tube produced by wrap() gets filled.
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel 64
 - OpenSCAD Version 2024.11.04. From pythonscad.org

**Library & Graphics card information**
OpenSCAD Version: 2024.11.04 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2024.11.04 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof*, input-driver-dbus, lazy-union*, vertex-object-renderers-indexing*, textmetrics, import-function, predictible-output*
[... 298 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

gsohler, 2024-11-16:
I see the same behavior, when I switch the backend renderer to old CGAL.
When I configure the backend to "Manifold" and restart, the shown error is gone and top of the cylinder is hollow.
Suppose its an CGAL bug ...

willywong, 2024-11-16:
> When I configure the backend to "Manifold" and restart, the shown error is gone and top of the cylinder is hollow.

Is this configurable in 2024.11.04? I don't see the toggle under Preferences > Features.

gsohler, 2024-11-16:
Upstream openscad moved its place quite recently.
Its now in Edit->Preferences-Advanced->Backend

willywong, 2024-11-16:
Works fine when I switched to Manifolds 👌 

I can close this. Thank you!

closed 2024-11-16 by willywong

gusohler, 2024-11-16:
Admittedly, wrap does create self intersected shapes, which is not 100%
nice, but Manifold can handle it ...
i should actually improve that in the future.

willywong, 2024-11-16:
It feels sufficiently functional for now 👌