gsohler/openscad #47: Using a dovetail tool before a roundover tool causes dovetail tool to be used when roundover should...

WillAdams, 2024-11-23:
Okay, I have a library:

[gcodepreview.zip](https://github.com/user-attachments/files/17877943/gcodepreview.zip)

which is working up to a point:

[gcodepreviewtemplate_working.zip](https://github.com/user-attachments/files/17877951/gcodepreviewtemplate_working.zip)

[image]

but when additional roundover cuts are added with different tools:

[gcodepreviewtemplate_working.zip](https://github.com/user-attachments/files/17877976/gcodepreviewtemplate_working.zip)

it seems that the dovetail tool #814 from line 219 is used:

[image]

(for the cuts to the right edge and along the back)

despite toolchanges to other tools on lines 234 and 241 --- the strange this is that if those two tool change lines are commented out, then the previous roundover tool is used:

[image]

Is there some setting which could influence this?

gsohler, 2024-11-23:
William, your code does not produce the same models in my place.

1st code creates one facet at the top perimeter
2nd code creates 4 facets at the top perimeter

I cant see a difference in the dovetails of your pictures.
maybe we are facing yet another caching issue ? 

I have a very hard time to understand the context of the problem, because i only have very simple GCODE knowledge.

python has a very ordered way to store variables .
maybe you mistaken global variables with class variables ?

WillAdams, 2024-11-23:
Could you upload an image of what you are seeing? Maybe that would be a clue?

I'm going to go through and remove all the global things and see if that helps.

WillAdams, 2024-11-23:
I wrote up a bit on the G-code and tooling here:

https://forum.makerforums.info/t/roundover-tooling-and-toolpaths-and-previews/91951/2

working to add this in gcodepreview as well.

WillAdams, 2024-11-23:
Figured it out --- things I had forgotten/failed to take into consideration:

 - the system is set up for only a single roundover tool
 - didn't get the definition of cylinder() correct at one point --- I've updated the wiki to reflect that
 - I don't think I have the tool definition for roundover tools right (suspect a mixup between diameter and radius --- looking into that now)

Thanks for your patience and looking into this!

closed 2024-11-23 by WillAdams

gsohler, 2024-11-23:
Hi William, sorry for being so late here, had an exciting day with my family. 
I am glad that the issue apparently resolved already, i was quite confident, that the problem is inside the python code, not the c code, but i had to few time to find out/to prove

Let me know if you still like to get the pictures.

WillAdams, 2024-11-23:
I have a working file now, but also one which hangs up --- here are all three (working/notworking, and necessary class file) in case this is interesting:

[gcodepreviewtemplate_working.zip](https://github.com/user-attachments/files/17883031/gcodepreviewtemplate_working.zip)

[gcodepreviewtemplate_notworking.zip](https://github.com/user-attachments/files/17883032/gcodepreviewtemplate_notworking.zip)

[gcodepreview.zip](https://github.com/user-attachments/files/17883036/gcodepreview.zip)

and here's what the working file looks like when it renders:

[image]

(the notworking one should look similar)

gsohler, 2024-11-23:
the working result looks very advanced!

Does it have a special purpose or are you just exciting the abilities of
your library ?

"hanging" became a new meaning about 2-3 month back as all the calculations
are now performed in a different thread now.
meaning that gui should never freeze again

if you press f5/f6 and don't see ANYTHING while the gui is still works, it
literally means, thet the engine is still calculating ...

WillAdams, 2024-11-23:
Thanks!

Not sure what's up with the "notworking" file --- I'll keep an eye out for it happening again.

The idea behind this file is to test each possibility as it was converted --- next up is keyhole tools.