gsohler/openscad #48: Using Python classes from OpenSCAD

WillAdams, 2024-11-30:
Once one makes a class, e.g., gcodepreview from:

https://github.com/WillAdams/gcodepreview/blob/main/gcodepreview.py

and if one adds the line:

    from gcodepreview import *

at the bottom of the class file and then tries:

    gcp = gcodepreview(Base_filename,
               true, 
               true, 
               stockXwidth,
               stockYheight,
               stockZthickness,
               zeroheight,
               stockzero,
               retractheight,
               large_square_tool_num,
               toolradius,
               plunge,
               feed,
               speed);

the error:

    WARNING: Ignoring unknown function 'gcodepreview' in file gcodepreviewtemplate_0_7.scad, line 81

happens, which seems to indicate this is not (yet) supported. Would it be possible to do so?

gsohler, 2024-11-30:
this code works without any warning, after fixing the obvious  typo erros

from gcodepreview import *

Base_filename="basename"
gcp = gcodepreview(Base_filename,True,
           True, 100, 100,100,
           1, 2, 2, 1, 3,1, 50, 50)

WillAdams, 2024-11-30:
If I make test.scad:

    use <gcodepreview.py>
    
    //from gcodepreview import *
    
    Base_filename="basename"
    gcp = gcodepreview(Base_filename,True,
    True, 100, 100,100,
    1, 2, 2, 1, 3,1, 50, 50) 

I get:

    [ERROR: Parser error: syntax error in file ../../../Desktop/Untitled.scad, line 6](https://github.com/gsohler/openscad/issues/6,C:/Users/willa/OneDrive/Desktop/Untitled.scad)
    Execution aborted

in 2024.11.04

[image]

gsohler, 2024-11-30:
> Hi William,
>
> there are several issues with your code
>
> 1) OpenSCAD language requires you to put a semicolon after each
> statement,except for use and include, but there is none after:
>
> Base_filename="basename"
>
> 2) you choose openscad-function -> python-function mapping in your
> statement, but here is no gcodepreview function in your python file.
> instead there is a gcodepreview class in your python file
>
> Right now, pythonopenscad does not support  openscad-function ->
> python-class mapping, but of course I could probably make that work
> But next issue would be that you want to use member methods of your class,
> and openscad parser does not support sonstructions like
>
> value = variable . memberfunction(arguments)
>
> I could consider adding this next, that that would be  greater task.
> Long story short: it cannot work right now. maybe in the future.
>
> But dont deviate from your pretty python class, its way better than just
> functions with global variables.

I will investigate the effort to use python classes in openscad language.,
maybe even today.

OpenSCAD++ could be a killing feature 😂

gsohler, 2024-12-05:
let me know if OpenSCAD++ is working with gcodepreview or if there are some sub-features missing

WillAdams, 2024-12-05:
Thanks! Downloaded and installed --- will work through things this evening!

WillAdams, 2024-12-07:
Trying to implement this now.

Where should 

    from gcodepreview import *

go? 

I've tried it at the bottom of gcodepreview.py which seems to work --- the definition:

    module gcp(generatescad, generategcode, generatedxf) {
    gcp = gcodepreview(generatescad, 
                       generategcode, 
                       generatedxf, 
                       );
    }

works when called:

    gcp(false, false, false);

(presumably feeding the OpenSCAD "false" into Python which want "False" works)

but then I add:

    module popengcodefile(basefilename){gcp.opengcodefile(basefilename);}

    popengcodefile("export");

and get:

>WARNING: Ignoring unknown variable 'gcp'

[gcodepreview.zip](https://github.com/user-attachments/files/18045728/gcodepreview.zip)

Also tried:

    generatescad = false;
    generategcode = true;
    generatedxf = false;
    
    gcp(generatescad, generategcode, generatedxf);

but still get

>WARNING: Ignoring unknown variable 'gcp'

Is there something special which should be done in the class file?

I think I got all the semi-colons in the right place... will try a very simple file and if I can get it to work will build up from there to see if I can identify some sort of specifics.

WillAdams, 2024-12-07:
Once I figured out that I needed to provide complete module calls things seem to have begun working --- will iterate and post about any real problems as they come up.

WillAdams, 2024-12-07:
I previously had a lengthy post detailing a problem, which I deleted because I thought it was fixed, then I noticed that the ``fix'' was brought about by a typo.

Here is the current state of things/my questions.

First --- where should the line:

    from gcodepreview import *

go? I currently have it at the bottom of gcodepreview.py:

[gcodepreview.zip](https://github.com/user-attachments/files/18049388/gcodepreview.zip)

but I don't like that. Including it in the .scad file doesn't work --- perhaps in a second .py file which is also loaded via use?

Second, I have the file:

    use <gcodepreview.py>
    
    module gcp(generatescad, generategcode, generatedxf) {
    gcp = gcodepreview(generatescad, 
                       generategcode, 
                       generatedxf, 
                       );
    }
    
    generatescad = true;
    generategcode = true;
    generatedxf = false;
    
    gcp(generatescad, generategcode, generatedxf);
    
    currenttoolnum = 102;
    toolradius = 1.5875; 
    plunge = 100;
    feed = 400;
    speed = 10000;
    
    module popengcodefile(Base_filename, currenttoolnum, toolradius, plunge, feed, speed) {
    //gcp.
    opengcodefile(basefilename, currenttoolnum, toolradius, plunge, feed, speed);
    }
    
    popengcodefile("export", currenttoolnum, toolradius, plunge, feed, speed);

which loads the above and runs without error --- until I uncomment the gcp. line and bring up the following line so that it reads:

    gcp.opengcodefile(basefilename, currenttoolnum, toolradius, plunge, feed, speed);

and if I run things then I get:

>WARNING: Ignoring unknown variable 'gcp'

Could you post a minimal working skeleton file which I can try expanding on?

WillAdams, 2024-12-07:
Third time is the charm.

I believe I have a working version.

First, it doesn't seem necessary to have the line:

    from gcodepreview import *

I had it at the bottom of gcodepreview.py, but didn't like it there. Am I correct in leaving it out? If it needs to be included would it be okay to have it as a single line in a "gcp.py" file which is included via use <gcp.py>?

Second, it seems that wrapping up the initial class invocation in a module doesn't work --- it has to be at the top-level, so what I have thus far, and what seems to be working is:

    use <gcodepreview.py>
    
    generatescad = true;
    generategcode = true;
    generatedxf = false;
    
    gcp = gcodepreview(generatescad, 
                       generategcode, 
                       generatedxf, 
                       );
    
    currenttoolnum = 102;
    toolradius = 1.5875; 
    plunge = 100;
    feed = 400;
    speed = 10000;
    
    module popengcodefile(Base_filename, currenttoolnum, toolradius, plunge, feed, speed) {
        gcp.opengcodefile(Base_filename, currenttoolnum, toolradius, plunge, feed, speed);
    }
    
    popengcodefile("export", currenttoolnum, toolradius, plunge, feed, speed);

I'll keep adding to that an hopefully will get to a working tool, or if I have a "real" bug report, it will be something which actually makes sense.

gsohler, 2024-12-07:
> I'll keep adding to that and hopefully will get to a working tool, or if I

sorry, very busy with my family,

also working version here:

[image: image.png]

code is
:
use <gcodepreview.py>

gcp=gcodepreview();
echo(gcp);

gcp.opengcodefile(basefilename="gcode");
gcp.dxfline(1,2,3,4,5);
//gcp.closegcodefile();
cube();
-----------------
watch [pythonclass] in the log for proof
just the closegcodefile might have some weird "side" effect

yesterday i was convinced, that I could simply use BOSL2 in inside python
by translating BOSL2 code into python and its
actually pretty easy!
However, BOSL2 is *really* very advanced code, it uses all magic, openscad
can offer
so of course my translation failed.

worst thing is, that an openscad module can be called with variable length
argument list and variable length child length.
As python only has one argument list, its hard to map.

So I will progress with my original plan, to "osuse" a scad library.
However, the functions provided within need to be called with 2 argument
lists(one for arguments and one for children)

will progress with that late in the evening

Guenther

gsohler, 2024-12-07:
closegcode file is called fine.
closegcode itself fails because gc member variable is not initialized due
my call options.

in the file, which i received, this is still commented!
def __init__(self, # basefilename = "export", <-

Guenther

WillAdams, 2024-12-07:
Thanks! I think I managed to puzzle this out.

I'm writing things up on the wiki next, then will knock out the balance of the OpenSCAD wrapper.

William

WillAdams, 2024-12-08:
It looks as if there may be some minor issue with modules/functions which don't have parameters:

    module movetosafeZ(){
        gcp.movetosafeZ();
    }
    
    movetosafeZ();

results in:

>[WARNING: Python:: 'AttributeError("'gcodepreview' object has no attribute 'gc'")' in file gcodepreview3.scad, line 113](https://github.com/gsohler/openscad/issues/113,C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries/gcodepreview3.scad)

gsohler, 2024-12-08:
Willam,

I am quite sure, that there is an issue with your python class in that case.
you host a member variable 'gc'  and in this scenario you use it while
never setting it before.
maybe you should mask the statement in line 113 with  gnerategcode==True ?

WillAdams, 2024-12-08:
Good point.

There is a setting of the variable:

    self.gc = open(self.gcodefilename, "w")

in

    def opengcodefile(self, basefilename = "export",

I'll try breaking it down to first principles and test iteratively until I puzzle things out

closed 2024-12-08 by WillAdams