gsohler/openscad #49: How to determine if file handle exists/is open in class

WillAdams, 2024-12-09:
Okay, I have a class:

[gcodepreview.zip](https://github.com/user-attachments/files/18054729/gcodepreview.zip)

which has a working template file:

[gcodepreviewtemplate.zip](https://github.com/user-attachments/files/18054725/gcodepreviewtemplate.zip)

which has the line:

    print(gcp.gc.mode)

which when I run it thus nets "w" (twice?!?) which as I understand it, signifies that the file is writable.

But if I try to use the similar print(self.gc.mode) wrapped up in:

    def printgcmode(self):
        print(self.gc.mode)

I don't get anything as far as I can see in the console, and if I try:

    echo(gcp.gc.mode)

in a similar OpenSCAD file at what should be a matching location I get 

>ECHO: undef

and now I am getting:

>[WARNING: Python:: 'SystemError('<function gcodepreview.opengcodefile at 0x000002314AC31260> returned a result with an exception set')' in file gcodepreview3.scad, line 89](https://github.com/gsohler/openscad/issues/89,C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries/gcodepreview3.scad)
>[WARNING: Python:: 'SystemError('<function gcodepreview.opendxffile at 0x000002314AC31080> returned a result with an exception set')' in file gcodepreview3.scad, line 95](https://github.com/gsohler/openscad/issues/95,C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries/gcodepreview3.scad)
>[WARNING: Python:: 'SystemError('<function gcodepreview.opendxffiles at 0x000002314AC30F40> returned a result with an exception set')' in file gcodepreview3.scad, line 101](https://github.com/gsohler/openscad/issues/101,C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries/gcodepreview3.scad)
>[WARNING: Python:: 'SystemError("<class 'float'> returned a result with an exception set")' in file gcodepreview3.scad, line 107](https://github.com/gsohler/openscad/issues/107,C:/Users/willa/OneDrive/Documents/OpenSCAD/libraries/gcodepreview3.scad)

for the file:

[gcodepreview3.zip](https://github.com/user-attachments/files/18054771/gcodepreview3.zip)

Am I going about determining if the file object is open/available in the wrong way? 

Is there something about calling a file from a Python object from a class in Python in OpenPythonSCAD which doesn't work as it does for calling a file from a Python command in Python in OpenPythonSCAD?

WillAdams, 2024-12-09:
Simplifying this down I now have:

[gcfile.zip](https://github.com/user-attachments/files/18055047/gcfile.zip)

    #!/usr/bin/env python
    
    import sys
    
    class gcfile:
    
        def __init__(self, #basefilename = "export",
                     generatescad = False,
                     generategcode = False,
                     generatedxf = False,
                      ):
            self.generatescad = generatescad
            self.generategcode = generategcode
            self.generatedxf = generatedxf
            self.generatedxfs = False
    
        def opengcodefile(self, basefilename = "export",
                          currenttoolnum = 102,
                          toolradius = 3.175,
                          plunge = 400,
                          feed = 1600,
                          speed = 10000
                          ):
            self.basefilename = basefilename
            self.currenttoolnum = currenttoolnum
            self.toolradius = toolradius
            self.plunge = plunge
            self.feed = feed
            self.speed = speed
            if self.generategcode == True:
                self.gcodefilename = basefilename + ".nc"
                self.gc = open(self.gcodefilename, "w")
    
        def writegc(self, *arguments):
            line_to_write = ""
            for element in arguments:
                line_to_write += element
            self.gc.write(line_to_write)
            self.gc.write("\n")
    
        def closegcodefile(self):
            self.gc.close()

and a gcfiletemplate.py file:

[gcfiletemplate.zip](https://github.com/user-attachments/files/18055059/gcfiletemplate.zip)

    #!/usr/bin/env python
    
    import sys
    
    from gcfile import *
    
    # [Export] */
    Base_filename = "aexport"
    
    gcf = gcfile(True, #generatescad
                 True, #generategcode
                 True, #generatedxf
                 )
    
    gcf.opengcodefile(Base_filename)
    
    gcf.writegc("Test")
    
    gcf.closegcodefile()
 
which successfully generated "aexport.nc" with the content "Test".

Then we created:

[gcfiletest.zip](https://github.com/user-attachments/files/18055075/gcfiletest.zip)

    use <gcfile.py>
    
    // [Export] */
    Base_filename = "aexport";
    
    gcf = gcfile(true, 
                 true, 
                 true, 
                 );
    
    module opengcodefile(basefilename, currenttoolnum, toolradius, plunge, feed, speed) {
        gcf.opengcodefile(basefilename, currenttoolnum, toolradius, plunge, feed, speed);
    }
    
    opengcodefile(Base_filename, 102, 3.175, 400, 1600, 10000);
    
    module writegc(text) {
        gcf.writegc(text);
    }
    
    writegc("Test");
    
    module closegcodefile() {
        gcf.closegcodefile();
    }
    
    closegcodefile();
    

which when run as the errors:

>[WARNING: Python:: 'AttributeError("'gcfile' object has no attribute 'gc'")' in file gcfiletest.scad, line 18](https://github.com/gsohler/openscad/issues/18,C:/Users/willa/OneDrive/Documents/GitHub/gcodepreview/gcfiletest.scad)
>[WARNING: Python:: 'AttributeError("'gcfile' object has no attribute 'gc'")' in file gcfiletest.scad, line 24](https://github.com/gsohler/openscad/issues/24,C:/Users/willa/OneDrive/Documents/GitHub/gcodepreview/gcfiletest.scad)

which I believe circles back to what seems to be a problem where a Python class object called from OpenSCAD isn't able to hold a file open.

If there's some other command or test which should be tried, I would be glad of it.

WillAdams, 2024-12-09:
Presumably a work-around here is to store the intended file contents in a variable and then open, write, and close in a single context.

Will look into that this evening.

gsohler, 2024-12-09:
thank you for the reduced testcase, very appreciated, will look into it asap,

gsohler, 2024-12-09:
Hi William,
Yes, error confirmed, but its a simple one, unrelatated to any files, which literally hurts you.

Its the simple fact, that openscad cannot yet pass true or false as arguments to python, just numbers and strings yet.

I have simplified your scad code like
''' 
use <gcfile.py>

// [Export] */
Base_filename = "aexport";

gcf = gcfile(1, 1,  1,        );
echo(gcf);   

gcf.opengcodefile(Base_filename, 102, 3.175, 400, 1600, 10000);
gcf.writegc("Test");
gcf.closegcodefile();
'''

matching python class is

'''
  def __init__(self, #basefilename = "export",
                 generatescad = 0,
                 generategcode = 0,
                 generatedxf = 0,
                  ):
        self.generatescad = False
        self.generategcode = False
        self.generatedxf = False

        if generatescad > 0:
            self.generatescad = True
        if generategcode > 0:
            self.generategcode = True
        if generatedxf > 0:
            self.generatedxf = True
'''

The  "missing feature" in pythonscad is easily added, just not "right now" but maybe matter of few hours.
So you can decide whether to change your arguments of wait for the fix.

gsohler, 2024-12-09:
this was the missing code , simple and the same time leaving a big hole behind

      case Value::Type::BOOL:
	return val.toBool()?Py_True:Py_False;

closed 2024-12-09 by WillAdams

WillAdams, 2024-12-09:
Great! Looking forward to the new version.