gsohler/openscad #50: Using a Python variable multiple times from OpenSCAD crashes

WillAdams, 2024-12-11:
Having a module:

    module setuptool(){
        toolpaths = currenttool();
    }
    
    setuptool();

works, but if I then add something like:

    module movetosafeZ(){
        toolpaths = toolpaths.union(gcp.movetosafeZ());
    }
    
    movetosafeZ();

then things crash. (toolpathtests = toolpaths.union(...); also crashed)

using the .py library: 

[gcodepreview.zip](https://github.com/user-attachments/files/18089234/gcodepreview.zip)

and working from the file:

[gcodepreview3.zip](https://github.com/user-attachments/files/18089242/gcodepreview3.zip)

gsohler, 2024-12-11:
I will look into it as I want to disable any crash.

However when looking at that from OpenSCAD perspective ONLY ,
I am wondering, that OpenSCAD will actually do.
You have an function and a variable assinment inside a module and my my understanding this is only valid/existrs  inside the module scope but not outside the module, doubt that is your intent

In last testcase with the bool, you have also wrapped everything into modules (i believe , that you think its neccessary), but I dont see a need for that.
Is there a reason, why you wrap everything in modules instead of writing itr plain ?

will report more information oncei i tested ...

WillAdams, 2024-12-11:
The file:

https://github.com/user-attachments/files/18089242/gcodepreview3.zip

should be the version which crashes.

I guess this all circles back around to the discussion at:

https://old.reddit.com/r/OpenPythonSCAD/comments/1h9ln7m/programming_styles_in_openpythonscad/

For now, I guess the "correct" thing to do would be to manage the 3D objects getting stuffed into a variable in Python, then wrap those commands in OpenSCAD.

Am I correct to assume that output(variable_containing_3D_object) as a Python command called from OpenSCAD will work?

Maybe a better way to describe this problem is:

 - a Python command returning a variable and assigning that to an OpenSCAD variable seems to work the first time --- should that be an error?

gsohler, 2024-12-11:
Just had time to test your testcase. It does not crash in my case(pythonscad stays open) but gives an error like:

WARNING: Ignoring unknown variable 'toolpaths' in file gcodepreview3.scad, line 119

Of course setuptool() moduleinstance is called and toolpaths gets assigned.
But as toolspaths goes out of scope after module in instaniated, toolpaths does not exist anymore(This is openscad feature)

if you invoke movetosafeZ module afterwartds, toolpaths is not defined anymore for next operation

The answer to your output question:
yes, if you invoke output() module in SCAD, it will find output function in python world.
But whatever Value  OpenSCAD can provide as argument, output() will not understand/accept it as it reuires a solid.

Yes, it will invoke, but will always return an error, i just tested

A Python function returning a variable is absoluely fine IMHO.
return solid to an module intanciation will display some thing, return a number to an assignment will assign the variable.

Hope that helps

WillAdams, 2024-12-12:
My understanding here:

 - the current working state in Python is that adding cuts to a toolpaths variable is manually handled by manipulating the variables directly with toolpaths.union() commands --- this works, but is clunky
 - it does not work to pass a 3D object from Python into a variable in OpenSCAD because variables in OpenSCAD are not set up to hold 3D models

My current plan is to write some new Python defs which will completely hide the stock and toolpaths variables --- that should then allow wrapping the new commands in OpenSCAD code and using them --- hopefully this weekend.

Since you're raising an error for this now, I think that addresses things, so I'm going to close this.

closed 2024-12-12 by WillAdams