gsohler/openscad #51: tooling is rough

ipsod, 2024-12-14:
I'm working on dev tooling.  As a Python developer, the initial experience is pretty jarring, because I just keep finding things I know how to do in every Python project I use but yours.

- [x] Install a package through a virtual env
- [ ] Install a package through a virtual env in the standard way
- [x] Get code-intelligence in VSCode, vim, PyCharm, etc.
- [ ] Output text to my terminal (not the OpenSCAD console)
- [ ] Use standard REPL
- [ ] Use REPL of my choice
- [ ] Use a Jupyter notebook
- [ ] Use a debugger
- [ ] Use PyLibFive (it was in the .pyi file)

In the past, I've used PyBind11 and cppyy to run C++ code inside of Python.  Do you think it might be possible to reverse your architecture, and put the C inside of the Python?  Because, that would solve most of these problemscross off 3 pretty important items, and also improve the first item, in that it would in fact use that virtual environment, instead of just pointing to its libs.

ipsod, 2024-12-14:
To install a package through a virtual env.

Create virtualenv as normal.

Add this to your code:
`sys.path.append("/path/to/venv/lib/python3.12/site-packages")`

gsohler, 2024-12-14:
1st of all, i appreciate your effort!
I have tried to use pybind11 in the past, but i was not able to map many/all of the features to pybind11.
I believe if we put C code into python code, we could be able to ship as a python module,
but the big disadvantage, would be, that we would not be able to g++ link into pythonscad executable and thus not having access to internal structures

gsohler, 2024-12-14:
jupyter notpook would be absolutely cool. I think you would need to run pythonscad in command line mode, output stl and display it with  javascript ?

ipsod, 2024-12-14:
Well, we don't want to lose the advantages, or we could just stick with SolidPython, right?
For me, SolidPython's major advantage is that it actually is a Python library just like any other, instead of being basically a black-box application with an API that accepts Python code as an input - but it's like throwing that code over a wall, not like the (optionally) interactive language I'm used to.  On the other hand, Solid Python is just throwing code over a wall to OpenSCAD.

So, we have two halves of a solution, but, it doesn't seem to add up to 1 yet.

I don't know how I'd use Jupyter notebooks specifically for this - I just use them all over the place because some problems are just easier to get at through them.  I'm sure it's easy enough to output images from openscad and easy enough to have jupyter display them, though, but right now I just want to be able to troubleshoot and tinker there, and the normal OpenSCAD preview is fine for output.

ipsod, 2024-12-14:
What about a client/server architecture?  Something where, you build out a Python server that embodies the .pyi file that defines your application, and a Python client that talks to it.  Would probably lose some cycles to encode/decode, but I'd rather have a slower application (within reason) than the current developer experience.

gsohler, 2024-12-14:
If you have a client/server solution, its very easy that the server/client versions/abilities get out of sync, 
The additional time for a single communication with additionan encode/decode might be neglible, but i can think of model
creation scenarious which will calculate each voxel separately and the additional time will expand significantly. additional difficuly is that python in pytonscad already run in threads.
If you have python on top rather than C, its literally a different application Its like  solidpython2 and if you prefer this type of approach, I feel you want enhance on solidpython.
Again, i feel the interface between server and client  could be very volatile.
Today's case: i am almost ready anouncing that i can instanciate SCAD libraries(like BOSL2) into python, this interface does not only contain paramter passing and object returning, but also includes the abilities to pass python children with the scad modules - just today's example

ipsod, 2024-12-14:
Yeah, that sounds tough.  Do you know of other ways to solve any of the problems I listed?

Not really sure what I wan't.  I got frustrated with PythonSCAD a year or two back and have basically focused on robotics motion planning and control for the past year or two while waiting for the situation to get better.  Was looking at other libraries, particularly PyMadCAD and CADQuery, when I left.  I was hoping that this might be a way back that wouldn't require so much re-learning (and giving up the benefits that CSG gives over brep or whatever).

I like where you're going.  I don't mean to discount your work in any way - I'm a fan.  I just get a very claustrophobic feeling from all the Python tools I'm used to using being cut off.

ipsod, 2024-12-14:
Also, Python on top is a totally different?  Necessarily?

I mean, I can see how that's probably so... but, I did have the experience of... I used to write C libraries for Arduino, and then write Python libraries to talk to them, with a gnarly communication layer between.  But, now, I just import the C into Python, and run the same code on my PC as is running on the MCU, and everything got a lot simpler.  You're dealing with something probably at least ten orders of magnitude more complex, though, so maybe this might be totally irrelevant.

ipsod, 2024-12-14:
Well, what you're doing is probably more complex, but your exposed API, the .pyi file, is really pretty narrowly defined.  I can't help but feel that there's a simple-enough solution.

gsohler, 2024-12-14:
For some of your issues i have easy solutions,  for others i dont.
redirecting python errors to openscad console can easily be turned off(redirection was the tough task)
For debugging i use "print" method instead of breakpoints, I am wondering if there are scenarios, where pdb is better.

I understand, that you are diappointed in some way, also I am sorry for that, but please understand that i am almost alone on this project and i do it in my spare time because i "believe"  its a great thing. No way that will/can change the speed of direction/speed drastically. But of course  you can influence with PR's.

When i started to use python, it was just my favorite language, but when i reconsidered my choice later, i still concluded to use python. its most similar to openscad because of the named parameter arguments.
BTW python is not on top, but its sitting next to SCAD parser. For SolidPython2 its on top.

may be my  the out of sync error is on anther frequency/speed, here it fails qjuite permanent. If you still feel, that you want python on top, you can add an server/client interface in ADDITION as PR, i will very likely accept it, but issue is that the added functionaity needs to be in sync with the embedded python functions.

I know that my project might appear rough to others, but thats only because of my inabilities in certain topics. This is were i depend on/hope on/ the help/contribution of others.

ipsod, 2024-12-14:
Oh, certainly, I understand - I'm in exactly the same boat with all of my development, just doing what I can with what I have.  Again, I don't mean to demean your project in any way, I wouldn't be putting time into it if I didn't also think it was worthwhile.  It's not my intent to demean your project by listing problems, but hopefully to improve it.

I will keep it in mind as I work.  If I find it painful enough, I will look further into creating a client/server architecture addition.

Thank you for sharing your work.

ipsod, 2024-12-14:
Oh.  For debugging, you can do better than print with log.  Log seems like print with more steps, but, sometimes it's very useful to have the granular controls it gives.  Turning up logs on this file and down on that one, for example.

gsohler, 2024-12-14:
Yeah, i believe it could be a benefit controlling pythonscad from "outside",  not only from a python module. Especially i think, it could be cool, to control the viewing parameters like center of view, focus, direction ...
pdb would immediately work if we  managed to connect the stdin path from the GUI to the python core ...

ipsod, 2024-12-14:
Yeah... can you control those viewing parameters from PythonSCAD?  I don't see any functions mentioned for that, can we get one?

gsohler, 2024-12-14:
There some special variables like $vpr in OpenSCAD only yet which can be used to change the view during an animation.
Its not yet possible in PythonSCAD

But with  an internal server it will be even possible to control from outside.

gsohler, 2024-12-15:
I just had some brainstorming.
With PythonSCAD it would be rather easy to have an option where pythonscad would
behave exactly like a command line python interpreter(ipython). PDB would woulk out of the box, it would be a REPR.
(PLUS all the geomtrical functions would of course be available).
With some minor interface it would actually be quite easy to use it in jupyter. I would definitely support you on the pythonscad side.

ipsod, 2024-12-15:
For right now, I'm heading back to SolidPython.  Sorry, but I don't have time for as large of a project as it seems to be, and I've found that it's unusable for me in its current state.

When you get OpenSCAD libraries working, and, ideally, you get stubs set up for some libraries (gears and bosl2, at least - maybe @jeff-dh can help?)... I think that might be all it took to get it back to usable for me.

And... speaking of SolidPython... I wish one or the other of you would adopt the other's idioms  (ie: SolidPython uses +, you use |) so that my code could be used with either library.

jeff-dh, 2024-12-15:
> And... speaking of SolidPython... I wish one or the other of you would adopt the other's idioms (ie: SolidPython uses +, you use |) so that code could be used with either library. I think that + is probably more obvious to most people, and it's also easier to type.

That's not true, SolidPython supports both:

```python
>>> from solid2 import cube
>>> cube(1) | cube(2)
union() {
        cube(size = 1);
        cube(size = 2);
}
>>> cube(1) & cube(2)
intersection() {
        cube(size = 1);
        cube(size = 2);
}

```

gsohler, 2024-12-15:
When talking about CSG operations, we are actually talking about Boolean
operations on Voxels (or just 3D content)
and clearly operators for boolean are AND , OR , NOR, NAND, INV and so on

When mapping  void to 0 and solid to 1, Addition might work for Union to
some extent, but as soon as you add 1 + 1 = 2
for overlapping of solids, 2 is not really a correct state.

PythonSCAD used + for union some time in the past, but I got corrected by
the community
Nowadays, PythonSCAD uses + [vector] for translation and * [ scalar or
vector] for scaling, which is far more logical

it' s even possible to translate to several positions at once by
expressing  solid + [[ 1,0,0], [10,0,0]] e.g.

jeff-dh, 2024-12-15:
> import any python library in a PythonSCAD file by installing it in a virtual env (in the standard way)

start openscad from within the virtual environment and voila......

```Shell
jeff@n ~/foo $python -m venv testEnv
jeff@n ~/foo $source testEnv/bin/activate
(testEnv) jeff@nupsi ~/foo $pip install solidpython2
[...]
(testEnv) jeff@nupsi ~/foo $~/Downloads/openscad/build/openscad 

```

gsohler, 2024-12-16:
Yes, installation is really easy!
But Solidpython is always dependent on underlying OpenSCAD. without openscad SolidPython does not make so much sense.
Furhtermore you are stuck to the features, which OpenSCAD offers. PythonSCAD got the extended Ability to export STEP files of foldable posscript Files(Where you can 2D print and create your model with Paper). Any Python-OpenSCAD combo has the disadvantage, that dataflow is always from python to OpenSCAD.  
On of the Adavantages of PythonSCAD is that you can get the mesh points from your solid at any time with mesh() function , modify the points with python and feed them back with polyhedron method.

ipsod, 2024-12-16:
> Yes, installation is really easy!
> But...

I think you might be misunderstanding what you're looking at.  @jeff-dh was showing how to launch PythonSCAD from inside of a virtualenv, so that PythonSCAD has access to that virtualenv's libraries (solidpython2 in this case).

jeff-dh, 2024-12-16:
> Any Python-OpenSCAD combo has the disadvantage, that dataflow is always from python to OpenSCAD.

What's the disadvantage? Or the otherway around, what's the advantage?
Especially with the security concerns around.... without openscad -> python, openscad would remain secure...

ipsod, 2024-12-16:
> What's the disadvantage? Or the otherway around, what's the advantage?

Did you see this? https://github.com/jeff-dh/SolidPython/discussions/71#discussioncomment-11577907

gsohler, 2024-12-17:
> Did you see this? jeff-dh/SolidPython#71 (comment)

if you have both data directions you could

1st create a custom model/mesh
2nd retrieve the points and triangles from the mesh
3rd calculate fancy parameters like volume, center of mass, surface , other
stuff
4th modify the  points, scale/skew/stretch them in a non-linear way
5th assemble back the points with polyhedron

ipsod, 2024-12-17:
Ok, so, I want that, but, to me, it seems like there are 5% of use-cases where your library has an advantage over SolidPython, and in the other 95%, SolidPython, for me at least, has major advantages (all the things in the OP, plus it has bosl2 with type hints if you add solidpython_extensions).  And, I mostly say this just to try and reinforce the case that you're missing something important, because, at present, the only way I can get what I want (SolidPython's features + PythonSCAD's features) is to convince other people that it's worthwhile, because CAD is really a side thing for me, just a tool I use, and I can't afford the time to become expert-enough to develop these things.

jeff-dh, 2024-12-17:
On Tue, 17 Dec 2024 10:49:39 -0800
ipsod ***@***.***> wrote:

> it has bosl2 with type hints if you add solidpython_extensions

btw: did you try solid2.extensions.bosl2 lately? I guess it should also
have "type hints".... (it used to not have 'em some time ago)

ipsod, 2024-12-19:
> btw: did you try solid2.extensions.bosl2 lately?

Just tried it.  Works great.

petaflot, 2026-02-22:
> but especially 1) is result of internal architecture

I pretty much guessed that.. I'd have to look into the code to see how it's done and what can be done to work around that, but my health does not allow it ATM (computer time is too "expensive")

> use and display hinge.py example  somewhere deep in the tests

that sounds like a fun game :-P I want to try that

> might appear as an "oddball"

duh. wait till you see me XD

I can travel freely anywhere in CH, so around Konstanzsee would be cool

petaflot, 2026-02-24:
> Thx for the bug "h" report, try to insert ASAP

heh.. found another one (REAME.md):

`Pre-built binaries are available at https://www.pythonscad.org/downloads.php.`

should be 

`Pre-built binaries are available at https://www.pythonscad.org/downloads/ .`

cheers