gsohler/openscad #52: setting up code-intelligence for editors

ipsod, 2024-12-15:
I've attached a zipped file structure.
[pythonscad_stubs.tar.gz](https://github.com/user-attachments/files/18141883/pythonscad_stubs.tar.gz)

You should publish it on PyPi under pythonscad_stubs or something.

To enable editor support, then, people would just have to:

create a virrtualenv
pip install pythonscad_stubs  (inside of the virtualenv)
point editor to that venv

ipsod, 2024-12-15:
Also, it still needs work.  There are stubs for PyLibFive, and I left them because they were in your original stub file, but PyLibFive doesn't actually seem to work.

I'm just realizing I didn't add anything for __add__, etc., either.

gsohler, 2024-12-15:
PyLIbFive does work if you use a version with LibFive compiled in
There are issues with static constructors for libfive in windows build,
which make it crash even before anybody can see a window.
this is why windows installers dont ship with libfive for the moment.

gsohler, 2024-12-18:
this is my 1st attempt

https://pypi.org/project/pythonscad-stubs/0.0.1/