gsohler/openscad #53: ERROR: Compilation aborted by exception: basic_string::_M_construct null not valid

WillAdams, 2024-12-17:
Given the .py class file:

[gcodepreview.zip](https://github.com/user-attachments/files/18159063/gcodepreview.zip)

If one uses it via the file:

[gcodepreviewtemplate.zip](https://github.com/user-attachments/files/18159074/gcodepreviewtemplate.zip)

The first time, it works as expected:

[image]

and if one loads the .scad file:

[gcodepreview3.zip](https://github.com/user-attachments/files/18159094/gcodepreview3.zip)

which calls:

[gcodepreview4.zip](https://github.com/user-attachments/files/18159095/gcodepreview4.zip)

at the moment it isn't generating anything (previously it would generate just the stock)

[image]

but if you switch back to the original .py file you get:

[image]

>ERROR: Compilation aborted by exception: basic_string::_M_construct null not valid

Haven't had any luck creating a simple file from scratch which causes this problem....

gsohler, 2024-12-17:
[image]
I can reproduce. Hopefully I will be able to make some progress

WillAdams, 2024-12-17:
That's great to hear! I was _not_ having any luck making a small file which showed the problem.

I'm probably going to try a full-on re-write of all files involved _with_ unit tests for both .py and .scad over the holiday.... hoping that will get us over this final hurdle.

gsohler, 2024-12-17:
the error stroke is  far away from the error source. long story short.
After many many test runs, i know, that there are module parameter errors in gcodepreview3.scad and pythonscad just ignored them without any harm to run in next fatal error. 
Now i am working on displaying the error correctly, so you can do the rest.

closed 2025-04-20 by gsohler