gsohler/openscad #54: Keyboard shortcuts for incrementing/decrementing numbers and forcing a re-evaluation?

WillAdams, 2024-12-18:
Thought this was an interesting project/UI:

https://github.com/timonoko/openscad_tweaker_in_emacs

https://www.youtube.com/watch?v=eG5lhLYvihQ

basically you put the cursor after a number and then the F8/F7 keys will increment/decrement and force a render.

Is there a pair of keys which aren't used for shortcuts which would make sense to assign thus? Could this be adapted in such a way as to only re-render/re-draw the affected element(s)?

gsohler, 2024-12-18:
Its a cool Feature, but openSCAD has it already!

Happily it works as well with Python Code.Put your cursor to the correct digit, press and hold ALT key,
than press cursor up/down keys

Hope it works for you

WillAdams, 2024-12-18:
I did not know that!

closed 2024-12-18 by WillAdams