gsohler/openscad #55: Booleans called from Python classes are either Undef or don't appear

WillAdams, 2024-12-24:
If one has a class file which has:

    from openscad import *
    
    class opclstst:
    
        def __init__(self,outputscad):
            self.outputscad = outputscad
    
        def checkoutputscad(self):
            print(self.outputscad)

It works as expected from Python:

    from opclstst import *
    
    opct = opclstst(False)
    
    opct.checkoutputscad()

yields:

>False

(twice for some reason)

but if one defines a .scad file with:

    use <opclstst.py>
    
    opct = opclstst(false);
    
    module checkoutputscad(){
        opct.checkoutputscad();
    }
    
    checkoutputscad();

The entirety of the output (there is some other code for making 3D stuff) is:

>Compiling design (CSG Tree generation)...
>Rendering Polygon Mesh using Manifold...
>Geometries in cache: 15
>Geometry cache size in bytes: 16456
>CGAL Polyhedrons in cache: 5
>CGAL cache size in bytes: 0
>Total rendering time: 0:00:00.039
>   Top level object is a 3D object (manifold):
>   Status:     NoError
>   Genus:      0
>   Vertices:       19
>   Facets:         34
>Measurements:
>   Surface: 617.59
>   Volume: 499.80
>Rendering finished.

with no sign of "False"

At one point in experimenting I was getting "Undef" when trying to echo or print a Boolean in a class.

This is using 12.17 on Windows.

gsohler, 2024-12-24:
yeah, there is a 2 bugs - one one my side, so l will solve mine first ..

gsohler, 2024-12-24:
xmas edition is uploaded.

The correct way to display the "False" from SCAD is:
'''
use <opclstst.py>
opct = opclstst(true);
echo(opct.checkoutputscad());
'''

the Reason for the doubled result is, that pythonscad runs the code twice:
1) to get the parameters for cudstomizer
2) real run

your gcodepreview speed is hardly effected, because main time is spent by the csg operations, ,which come after the 2 python runs,

WillAdams, 2024-12-24:
What a great Christmas present. Thank you!

gsohler, 2024-12-24:
william, of course you need to return the value in the python class:
 def checkoutputscad(self):
        return self.outputscad

WillAdams, 2024-12-24:
Thanks!

I'll eventually puzzle this all out.

For the two runs maybe you could bracket the first with explanatory text?

>Beginning run to set customizer values
...
>Completed setting customizer values

gsohler, 2024-12-24:
great idea. just implemented. in 1st run i limit execution until
last add_parameter statement

WillAdams, 2024-12-24:
I am really having a difficult time with this.

Finally broke down and tested:

    class opclstst:
    
        def __init__(self,outputscad):
            if outputscad == 1:
                self.outputscad = True
            else:
                self.outputscad = False
    
        def checkoutputscad(self):
            print("Checking if...")
            if self.outputscad == True:
                print("ops is true")
            else:
                print("ops is false")
            return self.outputscad

which when tested with:

    use <opclstst.py>
    
    opct = opclstst(1);
    
    echo(opct.checkoutputscad());

yielded:

>Compiling design (CSG Tree generation)...
>ECHO: true
>Rendering Polygon Mesh using Manifold...
>WARNING: No top level geometry to render

so, the rendering of True (Python) as true (OpenSCAD) works to some degree --- but apparently "print" commands in Python don't work for OpenSCAD? (am I mis-remembering that they used to?)

Another consideration is it doesn't seem to be possible to wrap up a class creation command in an OpenSCAD module:

    module callopct(tf){
    if (tf == true) {
    opclststtrue();
    } else {
    opclststfalse();
    }}

at least I couldn't get a working result --- should it work? Is there some architectural reason it shouldn't? Or have I been studying Lisp too much?

WillAdams, 2024-12-25:
However, the construct:

    use <opclstst.py>
    
    setting = true;
    
    thesetting = setting == true ? 1 : 2;
    
    opct = opclstst(thesetting);
    
    echo(opct.checkoutputscad());

works as expected (even though it's the sort of thing we were trying to avoid).

gsohler, 2024-12-25:
William,
at least in my place your example works like i expected it to be

The warning: WARNING: No top level geometry to render
is correct as you dont specify any geometry when you hit F6.

The print's in python work but they appear in the terminal console as i
don't redirect them for into-python calls. right now they only work if
toplevel is python

this one:

module callopct(tf){
if (tf == true) {
opclststtrue();
} else {
opclststfalse();
}}

dont know exactly what you want to try as i dont know what opclsttrue shall do.

In case it shall set a variable: it will not work as you are fighting
against openscad scopes.

(believe you want to set variables from latest post)

outputting geometries instead inside the if will works fine instead

My goal is to integrate python into openscad as much as possible. but
i cannot change openscad\s core concepts because i want to stay 100%
scad compatible(and nested scopes is a very important one)

WillAdams, 2024-12-25:
Yeah, the warning thing makes sense.

The idea behind the wrapping code was that there would be separate modules:

    module opclststtrue(){
        gcp = opclst(true);
    }

but it doesn't seem to work to instantiate a Python class/object wrapped up in an OpenSCAD module --- the OpenSCAD scope thing makes that obvious why, so now I understand that and will stop.

I've got things working in Python again (had a weird bug where cutting out a circle resulted in a diamond in the center because it was connecting beginning and end) so I think things will make more sense once I have the OpenSCAD code working.

gsohler, 2024-12-25:
I think you could implement your intent with an openscad function instead,
which does not include an unintentional scope

function opclstssttrue()=opclst(true);

and you would need to call it like:

gcp=opclstssttrue();

calling without equal sign will not work as its a module instantiation again

WillAdams, 2024-12-25:
I believe I have worked out the underlying difficulty --- Onedrive isn't consistently synching/making files available.

[image]

(a few moments after that screengrab the contents of the two windows became consistent)

I'm going to take a break and come back to this.

WillAdams, 2024-12-29:
I believe between getting files to update, using the toggle to assign 1/0 and then parsing that, and being sure to use a function to return things, this should be manageable if I can just manage to get notes written up on the wiki in such a way that they make sense to future me.

I believe for error-handling we may need a special case of an improperly wrapped Python def and OpenSCAD module/function, but that's something quite different.

closed 2024-12-29 by WillAdams