gsohler/openscad #56: extrude can crash openscad

reticulatedpines, 2024-12-25:
Some extrudes, possibly those where the initial shape ends up inside itself, cause a crash due to failed assertion.

I'm using Linux, gsohler openscad built from commit 947ab57ff.

This demonstrates the problem:
```
import openscad as scad

# crash - doesn't like extrudes that end inside itself?
rod = scad.circle(5).path_extrude([[0, 0, 0], [0, 0, 20], [5, 0, 0]])

# same direction for last move but further distance is okay:
#rod = scad.circle(5).path_extrude([[0, 0, 0], [0, 0, 20], [50, 0, 0]])

# exact same path but smaller circle is okay:
#rod = scad.circle(0.05).path_extrude([[0, 0, 0], [0, 0, 20], [5, 0, 0]])

scad.output(rod)
```

CLI output on crash:
```
$ ../openscad/build/openscad crash_01.py 
openscad: [local_path]/openscad/src/geometry/GeometryEvaluator.cc:2954: virtual Response GeometryEvaluator::visit(State&, const PathExtrudeNode&): Assertion `extruded' failed.
Aborted
```

This is an immediate crash of the whole GUI.  Re-opening the file crashes again, so a user will need to fix the file with an external editor.  No feedback is given on what line in the file has the problem, so working out the required change could be annoying, especially if the path was computed.

gsohler, 2024-12-25:
Thank you for testing PythonSCAD.
Your testcase certainaly demands  extreme conditions, but indeed it should not crash. 
In any case i have converted into a warning message.
Just pull latest version to see the fix

reticulatedpines, 2024-12-26:
Thanks for the quick fix - I confirm this is fixed for me with 66f421940.

PythonSCAD is a lot easier for me to use than OpenSCAD, thanks for all the work getting it running.

gsohler, 2024-12-26:
Thank you for your feedback - wea re open to receive further feedback

closed 2024-12-26 by gsohler