gsohler/openscad #57: PythonSCAD can use stale python content

reticulatedpines, 2024-12-26:
If the file opened by PythonSCAD is edited and saved, this file is automatically reloaded (that is, any changes take effect).

If the open file imports from another python file, and *that* file is edited and saved, these changes are never visible.  PythonSCAD needs to be restarted before these changes take effect.

To demonstrate, make two files in the same directory.  One called stale_test.py:
```
import openscad as scad

import StaleTest

print(StaleTest.test)
```

One called StaleTest.py:
```
test = "hello"
```

Open stale_test.py in PythonSCAD, "hello" will be logged.  Edit StaleTest.py to change the string to "goodbye", save stale_test.py to trigger a reload by PythonSCAD: "hello" will still be logged, not "goodbye".

If your code is modular, and you want to change anything outside of the main file, this is quite inconvenient :)

Might be a side effect of original OpenSCAD "use" and "include" effectively copying all the content into one file before compiling?  I've only tested on Linux, and caching bugs are often fiddly to reproduce, so there is some chance this is a local problem only.

WillAdams, 2024-12-27:
Thank you for bringing this up --- it is a clear and lucid description of difficulties which I have had (and been pestering and tormenting @gsohler about/with). 

Hopefully it will help. In the meanwhile, I've been trying to discipline myself to quit and re-start OpenPythonSCAD whenever the behaviour is not as expected.

reticulatedpines, 2024-12-27:
Haha, I'm glad if it helps.  It was quite confusing to debug, since really I was working with a complicated 3rd party library.

I guess somewhere in initialisation a python interpreter must be started, and reloading the main file doesn't invalidate caches for the included libraries.  But there are many possible causes.   Caching is often annoying to investigate.

gsohler, 2024-12-27:
yes, thats exactly where my strategy starts, will need some time to
implement as I mainly code during night-time ;)

reticulatedpines, 2024-12-27:
Perhaps a nice glass of wine would help!  The workaround is 100% reliable and I always know when to restart now, so it's not too bad to deal with.

gsohler, 2024-12-27:
Haha, I enjoyed a bottle of wheat beer. Now i implemented afix which literally does
del sys.modules[modulename] on c level and it appears to be effective.
my main challenge was to circumvent a known bug with PyDict_GetItemString and to rule out system modules.
in any case, the change is pushed.

WillAdams, 2024-12-29:
Also, are these links from the wiki pertinent?

https://stackoverflow.com/questions/2918898/prevent-python-from-caching-the-imported-modules 

and 

https://old.reddit.com/r/OpenPythonSCAD/comments/1g1qnxc/anyone_else_trying_to_develop_python_libraries_in/

reticulatedpines, 2024-12-29:
Confirmed fixed for me, thanks.

gsohler, 2024-12-29:
great! . just feel free to close the issue when there are no further actions needed.
BTW Did you already find the "Share Design" item in the menus ?

reticulatedpines, 2025-01-02:
I had not seen Share Design, now I have.  It will likely be a while before I use it!  I've never used OpenSCAD of any flavour before, this is a learning project :)

It looks like I may end up porting a cycloidal gear library to PythonSCAD, if I end up with something useful I'll let you know.

closed 2025-01-02 by reticulatedpines