gsohler/openscad #58: 3D offset generates unexpected geometry

KlumpRasmus, 2025-01-01:
**Describe the bug**
When trying to generate an offset (larger) version of an object the result is unexpected.

**To Reproduce**
 Run the code below

**Expected behavior**
A larger version of the original object, enlarged by 2mm on all faces.

**Code reproducing the issue**
 from openscad import *
  
  fn = 100
  c = cube([15, 30, 10]) + [15, -15, 0]
  p = hull(c, cylinder(h=7, d=30))
  p2 = offset(p, 2)
  output(p2)

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**

 - OS:  Ubuntu 24.04.1 LTS
 - System: ThinkPad x13
 - OpenSCAD Version: own build 2025.01.01

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

gsohler, 2025-01-02:
1st guess is that offset has issues with 2 adjacent faces with very similar but not equal angle.
pythonscad will combine those faces by accident and the lost vertix would be the result of only 2 faces meeting in one point.
will try to addresss this ASAP. 
I am very confident that an instant way to fix that is to reduce your fn fir the moment

KlumpRasmus, 2025-01-05:
Great, thanks. Let me know if I can test anything out.