gsohler/openscad #59: mesh example from website crashes the program

Sod-Almighty, 2025-01-09:
The following code crashes Python OpenSCAD 2023.10.07 every single time on an Intel Mac:

```
from openscad import *

u=cube([1,1,1]) | cylinder(d=1,h=10)
pts, tris = u.mesh()
```

gsohler, 2025-01-09:
Probably you mean 2023.10.08

This bug has been fixed. .this code works with recent versions, i just tested again
Unfortunately i dont have access to an intel mac anymore.
Is it possible you compile from source or try another platform ?

Sod-Almighty, 2025-01-09:
[image]

I can _try_ to compile from source.......although my experience of compiling other people's sources on Mac has often been frustrating.

Sod-Almighty, 2025-01-09:
> try another platform

There are no releases. You suggest I try another platform as an alternative to compiling from source, but where are the newer binaries to download?

gsohler, 2025-01-09:
suppose you got your current version from

pythonscad.org/download.php

There are very recent windows installers and linux AppImages available.

Also Apple Silicon is more recent, I just don't have access to an intel mac
anymore.

Sod-Almighty, 2025-01-09:
Ah, I see. The Intel Mac download is _significantly_ older than the other downloads. Perhaps it might be a good idea to highlight this fact in some way, so people know they're gonna be a year behind if they choose the Mac download?

I will try compiling from source; and/or the Windows binary.

closed 2026-04-25 by github-actions[bot]