gsohler/openscad #60: nimport not implemented in 2025.01.02

WillAdams, 2025-01-19:
Finally got around to trying:

https://old.reddit.com/r/OpenPythonSCAD/comments/1g3s2j5/have_your_libraries_with_you_wherever_you_are/

but unfortunately, when I added the line:

    nimport("https://raw.githubusercontent.com/WillAdams/gcodepreview/refs/heads/main/gcodepreview.py")

I got:

>ERROR: Traceback (most recent call last):
>  File "<string>", line 1, in <module>
>NameError: name 'nimport' is not defined

Did I get the syntax or location or usage wrong?

Just out of curiousity --- did you consider pushing this up to the main Python fork? Seems an _awesome_ option.

gsohler, 2025-01-20:
Thats great news!

nimport *does* work when using it from the correct place :) :)
nimport is placed next to all other openscad primitives as openscad.nimport

So just impor openscad first and it should work.
Admittedly, it does not work 100%. Once nimport downloaded a file from the web, on next start, the file cannot be re-downloaded thus overwritten due to some strange security mechanism until manually deleted.

Nope , I did not consider pushing this upstream

* syntrax would be different , it would be nimport <url>  without quotes and parantheis
* after all, it would be huge security risk, people could add malicious parts into the downloaded code without anybody notice

WillAdams, 2025-01-20:
Ah, that makes sense.

Naturally:

    from openscad import *
    nimport("https://raw.githubusercontent.com/WillAdams/gcodepreview/refs/heads/main/gcodepreview.py")
    from gcodepreview import *

works as expected --- going to go write this up now.

gsohler, 2025-01-20:
i think 3rd line is not needed. its done by 2nd

WillAdams, 2025-01-20:
>i think 3rd line is not needed. its done by 2nd 

and you are of course correct!

Updating things now.

William

wrote:

closed 2025-01-24 by gsohler