gsohler/openscad #61: Unable to show multiple shapes from dependency .scad imported with osinclude()

willywong, 2025-01-22:
**Describe the bug**
osinclude() works! It can call multiple modules/functions and return shapes.

HOWEVER if I do `show([shape1_from_osinclude, shape2_from_osinclude])`, it will ONLY show the last shape.

**To Reproduce**
```py
from openscad import *

HANDLEBAR_DIAM = 23    # ld
CLAMP_WIDTH = 20       # bh
CLAMP_THICKNESS = 5    # ws
GOPRO_CONNECTOR_VALUES_UNUSED = [15, 1, 2.8, 3.4, 3.6]

def im():
    ring = osinclude("mod_handlebar_mount_cust.scad")
    
    dims = [HANDLEBAR_DIAM, CLAMP_WIDTH, CLAMP_THICKNESS, *GOPRO_CONNECTOR_VALUES_UNUSED]
    
    # Bottom of the handle bar clamp
    bot = ring.LenkerVerbinder(False, dims=dims)

    # Top of the handle bar clamp, with a riser block.
    top = ring.LenkerVerbinder(True, dims=dims)
    return [top, bot]

x, y = im()

# This works as expected.
#show([x])

# This also works as expected.
#show([y])

# Expected: Show both pieces, with the last element raised up on the Z-axis for visibility.
# Actual: This only shows the last element in the array.
show([x, y.up(50)])
```

You'd need to grab the dependency `mod_handlebar_mount_cust.scad`. I pushed it to thingiverse over here https://www.thingiverse.com/thing:6922523.

**Expected behavior**

Expected: Show both pieces, with the last element raised up on the Z-axis for visibility.

Actual: This only shows the last element in the array.

See last 3 lines of the repro code above.

**Code reproducing the issue**

See above.

**Environment and Version info (please complete the following information):**
 - OS:Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2025.01.02, from pythonscad.org

**Library & Graphics card information**
```
OpenSCAD Version: 2025.01.02 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2025.01.02 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
[... 312 lines cut ...]
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control
```

willywong, 2025-01-22:
FYI this is not a critical bug. To workaround this, I can export STL one component at a time as

willywong, 2025-01-22:
Update: I found a strange way to have both items shown.

Full repro code below, but requires you uncomment the render() line after the first F5 to render.

```py
from openscad import *

HANDLEBAR_DIAM = 23    # ld
CLAMP_WIDTH = 20       # bh
CLAMP_THICKNESS = 5    # ws
GOPRO_CONNECTOR_VALUES_UNUSED = [15, 1, 2.8, 3.4, 3.6]

def im():
    ring = osinclude("mod_handlebar_mount_cust.scad")
    
    dims = [HANDLEBAR_DIAM, CLAMP_WIDTH, CLAMP_THICKNESS, *GOPRO_CONNECTOR_VALUES_UNUSED]
    
    # Bottom of the handle bar clamp
    bot = ring.LenkerVerbinder(False, dims=dims)

    # Top of the handle bar clamp, with a riser block.
    top = ring.LenkerVerbinder(True, dims=dims)
    return [top, bot]

x, y = im()

# This works as expected.
#show([x])

# This also works as expected.
#show([y])

# UNCOMMENT THIS AFTER RUNNING SHOW FOR THE FIRST TIME
# From testing: You want to render the last element in the array argument to show. Then both items show up.
#render(y)

# Expected: Show both pieces, with the last element raised up on the Z-axis for visibility.
# Actual: This only shows the last element in the array.
show([x, y.up(50)])
```

gsohler, 2025-01-22:
Hmm, i have seen the bug for a short time, , but now it works

[image]

I think you need to return a tuple when you want to receive a tuple, but i am not exactly sure if that is really needed.
But my pythonscad has another bug. can you see it ?
(its only local here ...)

willywong, 2025-01-22:
I dont think the tuple/list unpacking is the problem. I am able to reproduce the same bug if I assign each element separately.

```py
from openscad import *

HANDLEBAR_DIAM = 23    # ld
CLAMP_WIDTH = 20       # bh
CLAMP_THICKNESS = 5    # ws
GOPRO_CONNECTOR_VALUES_UNUSED = [15, 1, 2.8, 3.4, 3.6]

def im():
    ring = osinclude("mod_handlebar_mount_cust.scad")
    
    dims = [HANDLEBAR_DIAM, CLAMP_WIDTH, CLAMP_THICKNESS, *GOPRO_CONNECTOR_VALUES_UNUSED]
    
    # Bottom of the handle bar clamp
    bot = ring.LenkerVerbinder(False, dims=dims)

    # Top of the handle bar clamp, with a riser block.
    top = ring.LenkerVerbinder(True, dims=dims)
    return [top, bot]

x = im()[0]
y = im()[1]

# This works as expected.
#show([x])

# This also works as expected.
#show([y])

# UNCOMMENT THIS AFTER RUNNING SHOW FOR THE FIRST TIME
# From testing: You want to render the last element in the array argument to show. Then both items show up.
#render(y)

# Expected: Show both pieces, with the last element raised up on the Z-axis for visibility.
# Actual: This only shows the last element in the array.
show([x, y.up(50)])
```

[image]

willywong, 2025-01-22:
Same code, but with the `render(y)` line uncommented now would show both pieces.

[image]

gsohler, 2025-01-23:
I am able to reproduce this on my side now, but windows only ...
It looks like a memory corruption issue(access to freed data) 
I will keep it in mind and fix when I get ideas , how to address this.
Lets stay with the "render" trick. This command indeed is not useless even though you did not collect the result an a variable ;)