gsohler/openscad #64: translate() removes a solid's existing handles. But the syntactic sugar positioning methods retains handles.

willywong, 2025-02-12:
**Describe the bug**
translate() removes a solid's handles, but up(), down(), left(), right(), front(), back() preserves them.

**To Reproduce**

This baseline code is fine.

Uncommenting the second-to-last show() would demonstrate the error where translate() removes a solid's existing handles.

```py
from openscad import *
import copy

# Clone a identity matrix constant.
IDENT = copy.deepcopy(cube(1).origin)

c = cube([3, 3, 3])

# Set original origin as c.orig
c.orig = IDENT

def option1(c):
    # Option 1: Using translate() removes any existing handles.
    c = c.translate([10, 10, 10])

    # Errors.
    print(c.orig)

    return c

def option2(c):
    # Option 2: Using syntactic sugar directional methods preserves existing handles.
    c = c.up(10).right(10).back(10)

    # No errors.
    print(c.orig)

    return c

# Uncomment this to see error.
#show(option1(c))

show(option2(c))
```

**Expected behavior**
Handles should be preserved even if translate() is applied.

**Code reproducing the issue**
See above.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version OpenSCAD 2025.01.02 from pythonscad.org

**Library & Graphics card information**
```
OpenSCAD Version: 2025.01.02 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2025.01.02 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Manifold version: 3.0.1
Qt version: 5.15.11
[... 311 lines cut ...]
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

```

pythonscad, 2025-02-12:
THANK you for your extensive bug report.
maintaing handles for translate() was working before, but it got messed bug when implemtting the  explode operator like

solid ^ [[0,10], [0,5], [-2,2]] # this copies the solid 8 times

Now handles are back in case there is only one copy (default)
everything is checked in the latest master at 

http://github.com/pythonscad/pythonscad/

thanks again!

gusohler, 2025-02-12:
> solid ^ [[0,10], [0,5], [-2,2]] # this copies the solid 8 times
>
> How does this operator work? I suppose it is some sort of matrix
> multiplication, but I can't wrap my head around it :D

nope, no matrix here
the solid is just translated with different values of x (0,10), y(0,5) and
z(-2,2)
see "camera"  demo in the homepage
right now preparing a new windows release with the fixes

petaflot, 2025-02-12:
> > solid ^ [[0,10], [0,5], [-2,2]] # this copies the solid 8 times
> 
> How does this operator work? I suppose it is some sort of matrix multiplication, but I can't wrap my head around it :D

the matrix you see is obviously a representation of an orthogonal ("cubic") grid of dimension 2x2x2, hence the 8 copies

willywong, 2025-02-12:
This is an interesting operator :)

I can see it being handy if I vector precompute a bunch of coordinates to replicate a solid irt to origin.