gsohler/openscad #66: divmatrix(matrix1, matrix2) does not use inverse

seeraven, 2025-03-01:
**Describe the bug**
First of all a big thank you for your tremendous work!

However, I discovered a bug in the `divmatrix` operation. It works as expected for PyOpenSCAD objects, but when used with a 4x4 matrix object it behaves exactly like `multmatrix`.

**To Reproduce**
See the code example below.

**Expected behavior**
The output of the `divmatrix(M1, M2)` function should be `M2^-1 * M1`, that is the inverse of the matrix M2 should be multiplied with the matrix `M1`.

**Code reproducing the issue**
```
from openscad import *

identity = [
    [1.0, 0.0, 0.0, 0.0],
    [0.0, 1.0, 0.0, 0.0],
    [0.0, 0.0, 1.0, 0.0],
    [0.0, 0.0, 0.0, 1.0],
]

m = rotz(up(identity, 10), 45)
print(f"Matrix M: {m}")

# M^-1 * M should be identity matrix
result = divmatrix(m, m)
print(f"M^-1 * M: {result}")

# But the result is actually the same as a simple matrix multiplication
result2 = multmatrix(m, m)
print(f"M * M:    {result2}")
```

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 22.04
 - System: Intel PC 64-bit
 - OpenSCAD Version 2025.02.25 built from source and installed as a debian package

**Library & Graphics card information**
```
OpenSCAD Version: 2025.02.25.nightly (git )
System information: Linux 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 Ubuntu 22.04.5 LTS 20 CPUs 62.61 GB RAM
User Agent: OpenSCAD/2025.02.25.nightly (git ) (Linux x86_64; Ubuntu 22.04.5 LTS)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
Python Version: Python3.10.12
OpenCSG version: OpenCSG 1.6.0
Clipper2 version: 1.4.0
Manifold version: 3.0.1
Qt version: 5.15.3
QScintilla version: 2.11.6
InputDrivers: 
GLib version: 2.72.4
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.11.1
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, skin, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /usr/bin
Documents Path: /home/clemens/.local/share
User Documents Path: /home/clemens/Documents
Resource Path: /usr/share/openscad-pyscad
[... 483 lines cut ...]
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

**Additional context**
The problem seems to be in the function `python_multmatrix_sub()` at line 1407. When a matrix is given as an object, a matrix multiplication is performed regardless of the `div` parameter (which indicates the inverse should be used).

gsohler, 2025-03-01:
Thank you for the bug report - very appreciated.
Pythonscad has moved repository in the meantime.
I have already found the bug and fixed it in

https://github.com/pythonscad/pythonscad
see 
https://github.com/pythonscad/pythonscad/commit/11c563aa63b92b5a41230ad23c1c8f4d11d67207

Is the Lake in your profile picture Lake of Constance by chance ?

I

seeraven, 2025-03-01:
Wow, that was quick! Thanks a lot. I can confirm that the fix works.

No, if I remember correctly that was the port of Barcelona - about 18 years ago. :-)

Greetings from Stuttgart!

closed 2025-04-20 by gsohler