gsohler/openscad #67: offset() applied to a face's linear_extrude() result in crashes

willywong, 2025-03-17:
**Describe the bug**
offset() applied to a face's linear_extrude() result in crashes.

It does require a sequence of steps:
1. Access the 2d-like face via faces() first.
2. Perform linear_extrude() on a face.
3. Apply an offset() to the linear_extruded face.

**To Reproduce**
See the following code:

```py
from openscad import *

# On OpenSCAD 2025.03.15 

c = cube(5)

shell = []
for f in c.faces():
    outer = f.linear_extrude(10)
    inner = outer.offset(r=-2)

    # Expect: Each modified face extrusion is a rectangular pillar that is hollow in the middle.    
    shell.append(outer - inner)
    
# Instead, PythonSCAD crashes.
show(shell)
```

Funny enough: If I make the offset r POSITIVE, I do not see any crashes.

**Expected behavior**
No crash. I would expect each extruded face to have a hollow interior.

**Code reproducing the issue**
See above.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2025.03.15. From pythonscad.org

**Library & Graphics card information**
```
OpenSCAD Version: 2025.03.15 (git )
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 15.97 GB RAM
User Agent: OpenSCAD/2025.03.15 (git ) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
Python Version: Python 3.11.5
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.4.0
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, skin*, lazy-union*, vertex-object-renderers-indexing*, textmetrics, import-function, predictible-output*
[... 300 lines cut ...]
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

```

willywong, 2025-03-17:
Quick follow up: I am able to work around this with bounding boxes :)

I can take the magnitude from min and max coordinates from the bounding box calculated from mesh(), and then use resize() to perform functionally offset() on 3d shape.

Happy to say, this is not a blocking issue since other PythonSCAD primitives are so powerful :)

gsohler, 2025-03-17:
The Problem was the faces() command, the faces were in an left-handed coordinate system but was not visible
now after quite some eigen matrixes got fixed here is the new result

[image]