gsohler/openscad #69: Build 2025-04-13 doesn't recognize comments as comments in the editor

WillAdams, 2025-04-18:
If one has a comment which has a keyword in it:

    #Below command makes up for path issue  

it is syntax-highlighted as if it was code:

[image]

gsohler, 2025-04-20:
I just opened another big design because i was trying to chase another bug.

[image]

Comment lines are well displayed in the codeview.
Instead your python code looks like colored with SCAD syntax: 
"import" (to load an stl from disk) is exactly colored blue in SCAD

I have then alternated loading scad and py files and all was ok in my place.

Maybe you just give short instrucionts on how to see the wrong colors.
maybe its as simple as opening any python in 2024-12-30 ?

WillAdams, 2025-04-20:
Some random problem in Windows I guess --- I can't get it to happen again. Closing.

closed 2025-04-20 by WillAdams