gsohler/openscad #70: Subtracting from osuse imported object does not render the object.

wiw-pub, 2025-04-18:
**Describe the bug**
This is a somewhat odd bug (and I have a questionable workaround).

When ubtracting from an osuse imported object, pythonscad does not render the object.

**To Reproduce**

(1) Put this vanilla openscad file into a folder, and name it `test.scad`

```
module test() {
    cube([1, 1, 1]);
}

test();
```

(2) Repro the behavior with this pythonscad script. Make sure this is saved in the same folder adjacent to `test.scad` to ensure `osuse()` can source the scad file successfully:

```py
from openscad import *

fn = 100

def osuse_scad():
    return osuse('test.scad')

# After osuse, call one of its modules to produce a solid.
# test() returns a simple cube([1, 1, 1]).
ous = osuse_scad()
unit_cube = ous.test()

# This works fine. It will correctly show the bigger cube having a piece "bitten off" by the unit_cube.
#bigger_cube = cube(2)
#result2 = bigger_cube - unit_cube
#show(result2)

# However, when doing a difference with lhs as the osuse object, it does not show anything.
# stdout will print "WARNING: No top level geometry to render".
smaller_cube = cube(0.5)
result = unit_cube - smaller_cube
show(result)

# Oddly enough: If you uncomment line 14-15 and hit F6, I see expected results.
```

**Expected behavior**
When doing `unit_cube - smaller_cube` and showing the result, I expect to see the unit_cube having a corner "bitten" off. However, pythonscad will say "WARNING: No top level geometry to render".

Oddly enough, if I do "wasteful operation" performing an inverse subtraction (taking a basic pythonscad object and subtract WITH the osuse object) prior to the render on `unit_cube - smaller_cube`, I get the expected result. This is quite an odd workaround I have discover.

**Code reproducing the issue**
See above.

**Screenshots**
Expected behavior WITH THE WORKAROUND mentioned above:

[image]

Bug behavior:

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version: OpenSCAD 2025.03.19  from pythonscad.org

**Library & Graphics card information**
```
[... 325 lines cut ...]
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

```

gsohler, 2025-04-19:
osuse is known to have a memory corruption issue.
I will need to allocate lotss of time and have good ideas to address it.

wiw-pub, 2025-04-19:
Any rough idea why the workaround works? Does it mean the memory corruption is inconsistent (sometimes it happens, sometimes it does not)?

gsohler, 2025-04-20:
yes, exactly, sometimes the memory is overwritten and you dont a get a toplevel geometry.
Only thing for sure: its in osuse, because without everything works predicable

gsohler, 2025-06-06:
There has been many improvements since this bug was issued.
With recent versions, this  bug appears fixed.

wiw-pub, 2025-06-06:
I just tested this on 2025.05.22 build, and the reverse no longer works.

```py
from openscad import *

fn = 100

def osuse_scad():
    return osuse('test.scad')

# After osuse, call one of its modules to produce a solid.
# test() returns a simple cube([1, 1, 1]).
ous = osuse_scad()
unit_cube = ous.test()

# This works by itself (if you uncomment the below lines).
#smaller_cube = cube(0.5)
#result = unit_cube - smaller_cube
#show(result)

# This no longer works.
# But, if you uncomment ln 15 and 16, then this renders appropriately.
bigger_cube = cube(2)
result2 = bigger_cube - unit_cube
show(result2)
```

gsohler, 2025-06-06:
on Jun 4 (two days back) i fixed another issue with osuse

osuse is specifically complicated because parsing scad in OpenSCAD works
just fine, but
in case of usose, the Context defined by the Interpreter is prone to go out
of context and thus the model gets deleted before it can be displayed.
which platform are you using ?

wiw-pub, 2025-06-06:
I am on Windows 11, amd64.

I can pull the 2025.06.05 build later today when I can, and test it out 👌

gsohler, 2025-06-06:
great, let me know

Thank you for helping me make python for openscad great - either way

wiw-pub, 2025-06-06:
Thank you for all the great work on PythonSCAD!

wiw-pub, 2025-06-06:
Just tested 2025.06.05 build. No longer repro 👌 

I will close this issue. Thank you!

closed 2025-06-06 by wiw-pub